USBGuard/usbguard #366: start usbguard.service with error  <Failed with result 'start-limit-hit'.>

TheBigFish, 2020-03-05:
When `systemctl start usbguard.service` I got this message:

```bash
# sudo systemctl start usbguard.service
# sudo systemctl status usbguard
● usbguard.service - USBGuard daemon
   Loaded: loaded (/lib/systemd/system/usbguard.service; disabled; vendor preset: enabled)
   Active: failed (Result: start-limit-hit) since 三 2020-03-04 03:53:25 CST; 8s ago
     Docs: man:usbguard-daemon(8)
  Process: 2521 ExecStart=/usr/sbin/usbguard-daemon -k -c /etc/usbguard/usbguard-daemon.conf (code=killed, signal=SYS)
 Main PID: 2521 (code=killed, signal=SYS)

systemd[1]: usbguard.service: Unit entered failed state.
systemd[1]: usbguard.service: Failed with result 'signal'.
systemd[1]: usbguard.service: Service hold-off time over, scheduling restart.
systemd[1]: Stopped USBGuard daemon.
systemd[1]: usbguard.service: Start request repeated too quickly.
systemd[1]: Failed to start USBGuard daemon.
systemd[1]: usbguard.service: Unit entered failed state.
systemd[1]: usbguard.service: Failed with result 'start-limit-hit'.
```
Then I use `usbguard-daemon` manually start daemon, it can works, and the instuctions such as  `usbguad block-device` also works too. But the rules from `rules.conf` have no effects, even I reset the daemon.
 
Thanks in advance!

genodeftest, 2020-03-05:
What's your version of usbguard and your operating system? How about the permissions in `/etc/usbguard`? If you are on a linux distro with SELinux, can you please show us the result of running `ls -alZ /etc/usbguard`? Also, what does `journalctl -b --unit usbguard` give you?

genodeftest, 2020-03-05:
@dkopecek, @tweksteen : Is there a place where you collect common troubleshooting info? Or should I open a wiki page for that?

Cropi, 2020-03-29:
Hello @TheBigFish,
As @genodeftest pointed out, there can be a number of reasons why you are experiencing this problem. However, I think it is due to permission inconsistency. 
Could you tell us what permissions does your ```/etc/usbguard/rules.conf``` file have and also who the owner/group is?

ZoltanFridrich, 2021-02-11:
I suspect that you might have the wrong permissions set on the rules.conf file. Please check if your rules.conf file has 600 permissions set. The reason why rules.conf does not have effect when you run usbguard-daemon is because you need to call it with option -c (path to your usbguard-daemon.conf where RulePath is set).