USBGuard/usbguard #367: How to deactivate (temporarily) usbguard?

l0f4r0, 2020-03-07:
Hi,

How can I deactivate usbguard temporarily please?
Indeed, I would like to make some tests and I don't want usbguard to interfere...

I would have thought that a simple
`sudo systemctl stop usbguard`
would make it, but I still see messages like
`Device is not authorized for usage`
with
`sudo journalctl -kb`
after unplugging/replugging my device...

Thanks in advance :)

genodeftest, 2020-03-07:
It looks like `RestoreControllerDeviceState` has the wrong value in your `usbguard-daemon.conf`. Try setting it to `true`, then reboot and stop usbguard using `systemctl stop usbguard` again. Please have a look at the security note in the config file. Does that help?

l0f4r0, 2020-03-08:
> It looks like `RestoreControllerDeviceState` has the wrong value in your `usbguard-daemon.conf`. Try setting it to `true`, then reboot and stop usbguard using `systemctl stop usbguard` again. Please have a look at the security note in the config file. Does that help?

Thanks but unfortunately it doesn't change anything, "device is not authorized for usage" message is still present.

kabo, 2020-03-28:
I had the exact same issue. Worked around it by running `sudo systemctl disable usbguard` and rebooting, doing my thing, then `sudo systemctl enable usbguard`, rebooting again.

l0f4r0, 2020-03-28:
> I had the exact same issue. Worked around it by running `sudo systemctl disable usbguard` and rebooting, doing my thing, then `sudo systemctl enable usbguard`, rebooting again.

Thanks for sharing but it's quite a constraining workaround ;)

arjenlentz, 2020-04-04:
How about usbguard-daemon on startup storing the controller device states, regardless of the config setting. Then we could add a usbguard commd to restore that state - in @kabo 's case, it is a legitimate issue (flashing a USB device, they often do weird things) and the user clearly understands the implications.
usbguard already runs on the concept that if you're root, you can change/disable whatever you want. This is not a chance from that paradigm, it just makes the workflow possible without a reboot.

When done from a GUI, the GUI could restore the "allow everything" state but keep a window open, which when closed re-enables normal operations. That way a GUI desktop user is assisted with re-enabling appropriate USB security while not being prevented to, for instance, do upgrade flash tasks.

l0f4r0, 2020-04-04:
@arjenlentz Well, isn't it supposed to be already dealt with by usbguard (maybe not "regardless of the config setting" tough)? Otherwise, why would there be a RestoreControllerDeviceState variable?
If so, there might be an issue with the functionality because setting RestoreControllerDeviceState to true didn't help me...

arjenlentz, 2020-04-05:
@l0f4r0 I understand what you're saying, but the issue there is usability when you actually need it, vs. the security aspects.

Let's assume that we care about security, otherwise why bother with usbguard...
Having RestoreControllerDeviceState set to True during normal operations is not a good idea, because if the daemon were to be tricked into shutting down (through a bug or another vector), everything would be open.

So, let's further assume that RestoreControllerDeviceState is set to False. Then, to be able to do as you describe, the following flow is required:
 - Change the setting to True
 - Reboot the system
 - Stop usbguard daemon
 - <do what you need to do>
 - Change setting to True
 - Reboot the system

Simply changing the setting on a running system to True and even restarting the daemon before stopping it will not reset the values to their original "open" state.  For that, a reboot is required, or new functionality in the daemon that has it remember the original state even if it doesn't store it. The latter can then be used for the new feature I described.

ZoltanFridrich, 2021-02-18:
Are you sure that this is caused by USBGuard? When I search the source code for the phrase 'for usage' I get no results.

l0f4r0, 2021-02-28:
@ZoltanFridrich : I think it's certainly related to USBGuard because this exact message disappears when people apply modifications on USBGuard.
See examples below:

- https://askubuntu.com/questions/1269595/usb-device-is-not-authorized-for-usage
- https://bbs.archlinux.org/viewtopic.php?id=242613
- https://bugzilla.redhat.com/show_bug.cgi?id=1789923

ZoltanFridrich, 2021-03-01:
The message itself `Device is not authorized for usage` is generated by the system and not by USBGuard. When you stop the USBGuard, devices will stay in the same authorization state as was before the USBGuard shutdown. You can either do what @arjenlentz suggested with the RestoreControllerDeviceState or you can simply do `usbguard allow-device block` before shutting the USBGuard. This will allow all currently blocked devices (to allow blocked children of blocked devices you will need to repeat the command) and after USBGuard shutdown they will stay in this state. After you again start USBGuard, the policy will be applied as normal.

genodeftest, 2021-06-03:
This could be done if you wrote a shell script or alias which does something like this:
```
usbguard set-parameter ImplicitPolicyTarget allow
sleep 10
usbguard set-parameter ImplicitPolicyTarget block
```

leoluk, 2021-06-20:
If you want to avoid a reboot, stop the usbguard service, then set all default policies to 1:

```
for i in /sys/bus/usb/devices/**/authorized_default; do echo 1 > $i; done
```

sup2069, 2024-10-16:
> The message itself `Device is not authorized for usage` is generated by the system and not by USBGuard. When you stop the USBGuard, devices will stay in the same authorization state as was before [...]

For me on Centos9 server, the `usb-storage` was blacklisted:

`[centos@localhost ~]$ grep -r 'usb-storage' /etc/modprobe.d/`
`/etc/modprobe.d/usb-storage.conf:install usb-storage /bin/false`
`/etc/modprobe.d/usb-storage.conf:blacklist usb-storage`

Editing the `usb-storage.conf` and committing out the lines. I ran rebuilt the initramfs with `dracut` and then rebooted.
`[centos@localhost ~]$ sudo nano /etc/modprobe.d/usb-storage.conf`
`[centos@localhost ~]$ sudo dracut --force`
`[centos@localhost ~]$ sudo reboot`

On reboot, the USB stick was recognized and I was able to mount with `usbguard` running.