USBGuard/usbguard #369: Question: Does this protect against BadUSB?

CommonLoon102, 2020-03-26:
Against this attack:
https://opensource.srlabs.de/projects/badusb/wiki

muelli, 2020-03-26:
well. it depends. Do you have a more precise question?
Note that per-interface does not work yet: https://github.com/USBGuard/usbguard/issues/67

CommonLoon102, 2020-03-26:
For example, you have a memory stick but `The device can also spoof a network card and change the computer’s DNS setting to redirect traffic.`. So the OS would see an external drive plus a network card, or keyboard or something else. But the user would only see that it is an external drive and works as an external drive.

muelli, 2020-03-26:
that sounds like per-interface authorisation. AFAIU, you cannot configure USBGuard to allow interfaces, only. A patch would be good, though! :)

closed 2020-03-27 by CommonLoon102

radosroka, 2020-03-27:
> that sounds like per-interface authorisation. AFAIU, you cannot configure USBGuard to allow interfaces, only. A patch would be good, though! :)

Actually, you can use ```match-all``` keyword to specify mask for interfaces.