USBGuard/usbguard #376: 'rule-applied' condition doesn't do anything

foteei8, 2020-04-26:
I have the following rule:

`allow with-interface 08:*:* if !rule-applied`

This is the only rule that allows USB mass storage devices. Now the way I understand the 'rule-applied' condition, I should only ever be able to have one working USB mass storage device attached, right?

However, that's not the case:

`1: allow with-interface 08:*:* if !rule-applied
        8: allow id 0930:6545 serial "C86000886148B1601A26DAEB" name "DataTraveler 2.0" hash "TNNrgc0vN6g+NO/KtLQe8tncibkz83NYUI7QCpMzCQ4=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "1-5" with-interface 08:06:50 with-connect-type "hotplug"
        9: allow id 0bda:0158 serial "20071114173400000" name "USB2.0-CRW" hash "EPMhPiOzZ0iDtqzICXpOcYKWKBi1joO1JzBc3+1Jx58=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "1-7" with-interface 08:06:50 with-connect-type "hotplug"`

I can unplug and re-insert the DataTraveler USB stick, and it will get allowed every time, despite there being already another device that was explicitely allowed by the same rule.

So I tried to see what the 'rule-evaluated' condition would do instead. And to my surprise, it simply gets parsed as 'rule-applied' as well, and has the same effect as shown above, namely no effect at all:

`#  usbguard-rule-parser -f /etc/usbguard/rules.conf 
INPUT: allow with-interface equals { 08:*:* } if !rule-evaluated
OUTPUT: allow with-interface 08:*:* if !rule-applied`

Is this not working as intended, or am I gravely misunderstanding what the expected result should be here?

radosroka, 2020-07-18:
Hello,

I think that your ruleset works as expected. Rule number 1 matched your first mass storage once and then you inserted the second mass storage, rule was evaluated as non matching. Ruleset evaluation continues and the 8th rule matches your second mass storage which is breaking evaluation loop.

foteei8, 2020-07-28:
Hi.

8: and 9: aren't rules in my ruleset, they're automatically numbered devices in the "list-rules -d" output. Unfortunately github removed the indentation of those lines from the original output, which makes it look confusing.

ZoltanFridrich, 2021-03-04:
@foteei8 rule-applied condition definitely does not work as you assume. I believe that rule-applied returns true if this rule have ever been applied to a device, it means that the particular device could since then been blocked.

Please, take a look at this issue #456. The solution I suggested might just be what you want.

mikenrafter, 2022-12-13:
Sorry to necro-bump, but this issue is still present.

Firstly, the parser should not see `rule-applied` and `rule-evaluated` as the same item. They are distinctly different.  
Secondly, `allow if !rule-applied` is equivalent to `allow`... Plugging in two otherwise blocked devices results in them both connected. Only the first one should be connected, and the second blocked. This is a major roadblock for my use-case and I would like to see this fixed.

hartwork, 2022-12-13:
@mikenrafter not to argue bugs away or something: did you see…
> Allow a keyboard-only USB device only if there isn’t already a USB device with a keyboard interface allowed
> ```
> allow with-interface one-of { 03:00:01 03:01:01 } if !allowed-matches(with-interface one-of { 03:00:01 03:01:01 })
> ```
at https://usbguard.github.io/documentation/rule-language.html ? It sounds similar. Can that example be adapted into a workaround for your case? Just thinking aloud.