USBGuard/usbguard #379: Transcend StoreJet

jeblad, 2020-05-04:
I used the rules from “Reject devices with suspicious combination of interfaces”, so perhaps it should be said something about USB devices acting as a bridge.

My Transcend external disk shows up as 
```
Bus 002 Device 004: ID 174c:5106 ASMedia Technology Inc. ASM1051 SATA 3Gb/s bridge
```
which is somewhat as expected, but as it has a hid-interface (a button) it was still a bit unexpected.

Messing around a bit more it seems like even a plain `reject` after a drop-through won't block it.

jeblad, 2020-05-04:
Running `sudo usbguard generate-policy` gives
```
allow id 174c:5106 serial "     WD-WX71A34N7897" name "StoreJet Transcend" hash "iNJR7ONTbLyQQzjxh4e2S/QE0kcnTIgJEBuqBuQMeWw=" parent-hash "rHWt/jgO05HDfpDAqGXNpulnEfTpTBRrou94s9M19p0=" with-interface 08:06:50
```

radosroka, 2020-07-18:
Have you tried to use rule/policy without hashes?

jeblad, 2020-07-29:
Not sure what you ask about… I tried to reject everything that wasn't explicitly allowed, but the Trancend was still allowed. I tried to generate a rule to verify if it was detected properly, and it was detected as a HD. The HID wasn't detected.