USBGuard/usbguard #380: User is not being granted access to usbguard

madonius, 2020-05-11:
As described above my user with non-root privileges can not interact with usbguard

*OS*: Archlinux
*USBGuard Version*: 0.7.5-2

## Configured IPC access control file location
```bash
> cat /etc/usbguard/usbguard-daemon.conf | grep IPCAccess
IPCAccessControlFiles=/etc/usbguard/IPCAccessControl.d/
```

## Output of `usbguard-daemon -d`
```
…
(W) Failed to load IPC access control file /etc/usbguard/IPCAccessControl.d//USER
…
```

## IPC Access control file's contents
```bash
> cat /etc/usbguard/IPCAccessControl.d//USER
Devices=modify listen list
Rules=modify list
Exceptions=listen
```

radosroka, 2020-07-18:
What about file's permissions?

madonius, 2020-07-18:
Even setting the file's permissions to 644 does not do the trick :/
```
[1595109458.582] (i) Loading IPC access control files at /etc/usbguard/IPCAccessControl.d/
[1595109458.582] (T) Utility.cpp@361/loadFiles: L: USER : /etc/usbguard/IPCAccessControl.d//USER
[1595109458.582] (i) Loading IPC access control file /etc/usbguard/IPCAccessControl.d//USER
[1595109458.582] (W) Failed to load IPC access control file /etc/usbguard/IPCAccessControl.d//USER
```

madonius, 2020-07-18:
It appears that this might be an issue with the file's formatting. I do not see the mistake :thinking: 

Running:
`strace /usr/bin/usbguard-daemon -d -k -c /etc/usbguard/usbguard-daemon.conf`
yields:
```
openat(AT_FDCWD, "/etc/usbguard/IPCAccessControl.d//USER", O_RDONLY) = 10
read(10, "Devices=modify listen list\nRules"..., 8191) = 63
futex(0x7f1b7bfc2080, FUTEX_WAKE_PRIVATE, 2147483647) = 0
close(10)                               = 0
write(2, "[", 1[)                        = 1
write(2, "1595109710.746", 141595109710.746)          = 14
write(2, "] ", 2] )                       = 2
write(2, "(W)", 3(W))                      = 3
write(2, " ", 1 )                        = 1
write(2, "Failed to load IPC access contro"..., 81Failed to load IPC access control file /etc/usbguard/IPCAccessControl.d//USER) = 81
write(2, "\n", 1
)                       = 1
close(9)                                = 0
```

radosroka, 2020-07-19:
Have you tried to manage these files with CLI ?

```usbguard add-user ...```

madonius, 2020-07-19:
Thanks, this generates a valid file 

I found the issue with parsing the file. The issue was that the _Rules_ section I wrote is wrong. It should be _Policy_.
Further, adding the user via `usbguard add-user` generates a file separated with commas
```
Devices=list,modify,listen
Policy=list,modify
Exceptions=listen
```
in contrast the documentation (man-page) states:
```
Devices=modify list listen
Policy=list
Exceptions=listen
```
Shall I open a ticket/submit an issue for that?

madonius, 2020-07-19:
Restarting the daemon with the new configuration still does not yield the expected results.
```
$ > usbguard list-devices
ERROR: IPC connect: service=usbguard: Permission denied
```

madonius, 2020-07-19:
The issue seems to be in the systemd service file's `CapabilityBoundingSet=` configuration.

radosroka, 2020-07-20:
> Thanks, this generates a valid file
> 
> I found the issue with parsing the file. The issue was that the _Rules_ section I wrote is wrong. It should be _Policy_.
> Further, adding the user via `usbguard add-user` generates a file separated with commas
> 
> ```
> Devices=list,modify,listen
> Policy=list,modify
> Exceptions=listen
> ```
> 
> in contrast the documentation, man-page, gives the following example:
> 
> ```
> Devices=modify list listen
> Policy=list
> Exceptions=listen
> ```
> 
> Shall I open a ticket/submit an issue for that?

Feel free to open it.

radosroka, 2020-07-20:
> The issue seems to be in the systemd service file's `CapabilityBoundingSet=` configuration.

Why is that so?

madonius, 2020-07-20:
> > The issue seems to be in the systemd service file's `CapabilityBoundingSet=` configuration.
> 
> Why is that so?

I do not know why that is, just that if I comment out the line, my user has the expected permissions.

radosroka, 2020-07-20:
> > > The issue seems to be in the systemd service file's `CapabilityBoundingSet=` configuration.
> > 
> > 
> > Why is that so?
> 
> I do not know why that is, just that if I comment out the line, my user has the expected permissions.

If you have older libqb there is need for DAC_OVERIDE capability. 
See https://github.com/ClusterLabs/libqb/issues/369 and https://github.com/ClusterLabs/libqb/pull/381.

genodeftest, 2020-07-23:
Also, have you checked SELinux?

madonius, 2020-07-24:
If I see this correctly, this issue should not affect me.
```
$ > pacman -Ss libqb
community/libqb 1.0.5-2 [installed]
$ > pacman -Ss usbguard
pacman -Ss usbguard 
extra/usbguard 0.7.5-4 [installed]
```

Weirdly enougth with a, up to date, Debian testing where it works I have:
usbguard: 0.7.8+ds-1+b1
libqb: 1.0.5-1
CapabilityBoundingSet: CAP_CHOWN CAP_FOWNER CAP_DAC_READ_SEARCH CAP_DAC_OVERRIDE

madonius, 2020-07-24:
@genodeftest SELinux is off

LucasParsy, 2020-08-05:
I have the same issue as with @madonius 
the default capabilities for the usbguard service are
`CapabilityBoundingSet: CAP_CHOWN CAP_FOWNER`
but they are not sufficent to take in account my user rules. (created with `usbguard add-user`)

adding the  `CAP_DAC_READ_SEARCH CAP_DAC_OVERRIDE` capabilities in  `/usr/lib/systemd/system/usbguard.service` or `/etc/systemd/system/usbguard.service.d/override.conf` does the trick.

ZoltanFridrich, 2021-02-12:
> Even setting the file's permissions to 644 does not do the trick :/

afaik files need to have 600 permissions not 644

henri, 2023-08-15:
Confirmed, editing the config file by hand and changing the comers to spaces and then restarting the service - solves this problem.