USBGuard/usbguard #381: Mitigate Thunderspy

shaqarava, 2020-05-13:
[Thunderspy](https://thunderspy.io/):

> We have found 7 vulnerabilities in Intel’s design and developed 9 realistic scenarios how these could be exploited by a malicious entity to get access to your system, past the defenses that Intel had set up for your protection.

Proof of concept vulnerability checker: [spycheck-linux](https://github.com/BjornRuytenberg/spycheck-linux)

Possible mitigation code: [spiblock](https://github.com/BjornRuytenberg/spiblock)

This is a request for comment on rolling mitigation code into usbguard, if possible.

muelli, 2020-05-13:
how does this affect USB and USBGuard?

shaqarava, 2020-05-13:
> how does this affect USB and USBGuard?

The vulnerability is affecting usb-c/thunderbolt.