USBGuard/usbguard #382: Doesn't launch on Ubuntu 18.04.4 ( installation not ok + not running )

teameck, 2020-05-16:
**INSTALLATION**

teameck@linux:~$ sudo apt install usbguard
[sudo] Mot de passe de teameck : 
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
Les NOUVEAUX paquets suivants seront installés :
  usbguard
0 mis à jour, 1 nouvellement installés, 0 à enlever et 0 non mis à jour.
Il est nécessaire de prendre 0 o/115 ko dans les archives.
Après cette opération, 392 ko d'espace disque supplémentaires seront utilisés.
Sélection du paquet usbguard précédemment désélectionné.
(Lecture de la base de données... 265054 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../usbguard_0.7.2+ds-1_amd64.deb ...
Dépaquetage de usbguard (0.7.2+ds-1) ...
Paramétrage de usbguard (0.7.2+ds-1) ...
Created symlink /etc/systemd/system/dbus-org.usbguard.service → /lib/systemd/system/usbguard-dbus.service.
Created symlink /etc/systemd/system/multi-user.target.wants/usbguard-dbus.service → /lib/systemd/system/usbguard-dbus.service.
Created symlink /etc/systemd/system/basic.target.wants/usbguard.service → /lib/systemd/system/usbguard.service.
A dependency job for usbguard-dbus.service failed. See 'journalctl -xe' for details.
Job for usbguard.service failed because the control process exited with error code.
See "systemctl status usbguard.service" and "journalctl -xe" for details.
Traitement des actions différées (« triggers ») pour dbus (1.12.2-1ubuntu1.1) ...
Traitement des actions différées (« triggers ») pour man-db (2.8.3-2ubuntu0.1) ... 

======================== **journalctl -xe** ==================================
teameck@linux:~$ journalctl -xe
-- Support: http://www.ubuntu.com/support
-- 
-- L'unité (unit) usbguard.service a commencé à démarrer.
mai 16 03:25:22 linux usbguard-daemon[10864]: [1589592322.961] (E) ERROR: Configuration: /etc/usbguard/rules.conf: :1:18: parse error matching usbguard::
mai 16 03:25:22 linux usbguard-daemon[10864]: ERROR: Configuration: /etc/usbguard/rules.conf: :1:18: parse error matching usbguard::RuleParser::rule
mai 16 03:25:22 linux systemd[1]: usbguard.service: Control process exited, code=exited status=1
mai 16 03:25:22 linux systemd[1]: usbguard.service: Failed with result 'exit-code'.
mai 16 03:25:22 linux systemd[1]: Failed to start USBGuard daemon.
-- Subject: L'unité (unit) usbguard.service a échoué
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- L'unité (unit) usbguard.service a échoué, avec le résultat RESULT.
mai 16 03:25:23 linux systemd[1]: usbguard.service: Service hold-off time over, scheduling restart.
mai 16 03:25:23 linux systemd[1]: usbguard.service: Scheduled restart job, restart counter is at 5.
-- Subject: Le redémarrage automatique d'une unité (unit) a été planifié
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Le redémarrage automatique de l'unité (unit) usbguard.service a été planifié, en
-- raison de sa configuration avec le paramètre Restart=.
mai 16 03:25:23 linux systemd[1]: Stopped USBGuard daemon.
-- Subject: L'unité (unit) usbguard.service a terminé son arrêt
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- L'unité (unit) usbguard.service a terminé son arrêt.
mai 16 03:25:23 linux systemd[1]: usbguard.service: Start request repeated too quickly.
mai 16 03:25:23 linux systemd[1]: usbguard.service: Failed with result 'exit-code'.
mai 16 03:25:23 linux systemd[1]: Failed to start USBGuard daemon.
-- Subject: L'unité (unit) usbguard.service a échoué
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- L'unité (unit) usbguard.service a échoué, avec le résultat RESULT.
mai 16 03:25:26 linux sudo[10567]: pam_unix(sudo:session): session closed for user root
lines 2332-2367/2367 (END)

**Service STATUS** 

teameck@linux:~$ sudo systemctl status usbguard
[sudo] Mot de passe de teameck : 
● usbguard.service - USBGuard daemon
   Loaded: loaded (/lib/systemd/system/usbguard.service; enabled; vendor preset:
   Active: failed (Result: exit-code) since Sat 2020-05-16 02:56:45 CEST; 21min 
     Docs: man:usbguard-daemon(8)
  Process: 6755 ExecStart=/usr/sbin/usbguard-daemon -f -s -c /etc/usbguard/usbgu

mai 16 02:56:45 linux systemd[1]: usbguard.service: Control process exited, code
mai 16 02:56:45 linux systemd[1]: usbguard.service: Failed with result 'exit-cod
mai 16 02:56:45 linux systemd[1]: Failed to start USBGuard daemon.
mai 16 02:56:45 linux systemd[1]: usbguard.service: Service hold-off time over, 
mai 16 02:56:45 linux systemd[1]: usbguard.service: Scheduled restart job, resta
mai 16 02:56:45 linux systemd[1]: Stopped USBGuard daemon.
mai 16 02:56:45 linux systemd[1]: usbguard.service: Start request repeated too q
mai 16 02:56:45 linux systemd[1]: usbguard.service: Failed with result 'exit-cod
mai 16 02:56:45 linux systemd[1]: Failed to start USBGuard daemon.

muelli, 2020-05-28:
ouf.
I believe that people are more inclined to look at your issue if you make sure it is presentable enough.
So for starters, can you run everything with LC_ALL=C? And the least you could do is to use triple backticks to not have the markdown thingy turn everything into a bold mess.

lemmy04, 2020-07-07:
Might be related to #392

lemmy04, 2020-07-07:
Hm. it works fine on Ubuntu 20, but not on Leap 15.2 or Ubuntu 18. Ubuntu 20 has systemd 245, Leap 15.2 has systemd 234, Ubuntu 18 has systemd 237... I'm starting to see a pattern here.

hartwork, 2022-02-05:
The important parts of [the log above](https://github.com/USBGuard/usbguard/issues/382#issue-619349603) seem to be:
```
mai 16 03:25:22 linux usbguard-daemon[10864]: [1589592322.961] (E) ERROR: Configuration: /etc/usbguard/rules.conf: :1:18: parse error matching usbguard::RuleParser::rule
mai 16 03:25:22 linux usbguard-daemon[10864]: ERROR: Configuration: /etc/usbguard/rules.conf: :1:18: parse error matching usbguard::RuleParser::rule
```