USBGuard/usbguard #389: new version 0.7.8 causes 'flapping' with usb-c dock

rfrohl, 2020-06-18:
Hi,
with the new version 0.7.8 I am seeing problems with the usb-c dock from Dell that I am using. Previously I was using 0.7.6 and there were occasional problems, but day to day normal operations was possible. 

From a users perspective there are repeat delays with mouse and keyboard usage. Networking over the dock is not usable anymore. 

From the logs it seems like usbguard is causing the usb-c dock to disconnect (log from journalctl -e):
```
kernel: usb 4-1: USB disconnect, device number 107
kernel: usb 4-1.2: USB disconnect, device number 108
kernel: xhci_hcd 0000:3b:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
kernel: r8152 4-1.2:1.0 enp59s0u1u2: Stop submitting intr, status -108
NetworkManager[2030]: <info>  [1592471887.3966] device (enp59s0u1u2): state change: unavailable -> unmanaged (reason 'removed', sys-iface-state: 'removed')
ModemManager[2001]: <info>  (net/enp59s0u1u2): released by device '/sys/devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:3b:00.0/usb4/4-1/4-1.2'
ModemManager[2001]: <info>  Couldn't check support for device '/sys/devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:3b:00.0/usb4/4-1/4-1.2': Operation was cancelled
usbguard-daemon[32509]: [1592471887.476] (A) uid=0 pid=32509 result='SUCCESS' device.rule='allow id 0bda:8153 serial "000002000000" name "USB 10/100/1000 LAN" hash "plV2tli5ayK5fvzmGw9DGYJoo0PFycigieW6uOc/sLc=" p
Mp/xEwGiAm8Bw+Wrs9KSGHvZqgPE=" via-port "4-1.2" with-interface { ff:ff:00 02:06:00 0a:00:00 0a:00:00 } with-connect-type "unknown"' device.system_name='/devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0
 type='Device.Remove'
usbguard-daemon[32509]: [1592471887.481] (A) uid=0 pid=32509 result='SUCCESS' device.rule='allow id 0424:5807 serial "" name "USB5807 Hub" hash "sNc63ffiWf8Y7NCMp/xEwGiAm8Bw+Wrs9KSGHvZqgPE=" parent-hash "JF1EpD/Z
XfMgHgag=" via-port "4-1" with-interface 09:00:00 with-connect-type "hotplug"' device.system_name='/devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:3b:00.0/usb4/4-1' type='Device.Remove'
kernel: usb 4-1: new SuperSpeed Gen 1 USB device number 109 using xhci_hcd
usbguard-daemon[32509]: [1592471887.772] (A) uid=0 pid=32509 result='SUCCESS' device.rule='block id 0424:5807 serial "" name "USB5807 Hub" hash "sNc63ffiWf8Y7NCMp/xEwGiAm8Bw+Wrs9KSGHvZqgPE=" parent-hash "JF1EpD/Z
XfMgHgag=" via-port "4-1" with-interface 09:00:00 with-connect-type "hotplug"' device.system_name='/devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:3b:00.0/usb4/4-1' type='Device.Insert'
kernel: usb 4-1: New USB device found, idVendor=0424, idProduct=5807, bcdDevice= 2.04
kernel: usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
kernel: usb 4-1: Product: USB5807 Hub
kernel: usb 4-1: Manufacturer: Microchip
kernel: usb 4-1: Device is not authorized for usage
kernel: hub 4-1:1.0: USB hub found
kernel: hub 4-1:1.0: 7 ports detected
kernel: usb 4-1: authorized to connect
usbguard-daemon[32509]: [1592471887.775] (A) uid=0 pid=32509 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:3b:00.0/usb4/4-1' target.new='allow' device.rule='
" name "USB5807 Hub" hash "sNc63ffiWf8Y7NCMp/xEwGiAm8Bw+Wrs9KSGHvZqgPE=" parent-hash "JF1EpD/Z/zPMsM9Z0zaZn4eerhHZzE8zfmOXfMgHgag=" via-port "4-1" with-interface 09:00:00 with-connect-type "hotplug"' target.old='
Update'
kernel: usb 4-1.2: new SuperSpeed Gen 1 USB device number 110 using xhci_hcd
usbguard-daemon[32509]: [1592471888.084] (A) uid=0 pid=32509 result='SUCCESS' device.rule='block id 0bda:8153 serial "000002000000" name "USB 10/100/1000 LAN" hash "plV2tli5ayK5fvzmGw9DGYJoo0PFycigieW6uOc/sLc=" p
Mp/xEwGiAm8Bw+Wrs9KSGHvZqgPE=" via-port "4-1.2" with-interface { ff:ff:00 02:06:00 0a:00:00 0a:00:00 } with-connect-type "unknown"' device.system_name='/devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0
 type='Device.Insert'
usbguard-daemon[32509]: [1592471888.086] (A) uid=0 pid=32509 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:3b:00.0/usb4/4-1/4-1.2' target.new='allow' device.
rial "000002000000" name "USB 10/100/1000 LAN" hash "plV2tli5ayK5fvzmGw9DGYJoo0PFycigieW6uOc/sLc=" parent-hash "sNc63ffiWf8Y7NCMp/xEwGiAm8Bw+Wrs9KSGHvZqgPE=" via-port "4-1.2" with-interface { ff:ff:00 02:06:00 0a
ect-type "unknown"' target.old='block' type='Policy.Device.Update'
kernel: usb 4-1.2: New USB device found, idVendor=0bda, idProduct=8153, bcdDevice=30.11
kernel: usb 4-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=6
kernel: usb 4-1.2: Product: USB 10/100/1000 LAN
kernel: usb 4-1.2: Manufacturer: Realtek
kernel: usb 4-1.2: SerialNumber: 000002000000
kernel: usb 4-1.2: Device is not authorized for usage
kernel: usb 4-1.2: authorized to connect
kernel: usb 4-1.2: Device not responding to setup address.
kernel: usb 4-1: USB disconnect, device number 109
kernel: usb 4-1.2: device not accepting address 110, error -22
kernel: usb 4-1.2: Dell TB16 Dock, disable RX aggregation
kernel: r8152 4-1.2:1.0 (unnamed net_device) (uninitialized): Using pass-thru MAC addr e4:b9:7a:c5:89:86
kernel: r8152 4-1.2:1.0 eth0: v1.11.11
kernel: usb 4-1.2: USB disconnect, device number 110
```

Turning off the usbguard daemon stops the flapping and normal use is possible again.

The firmware of the dock was not changed recently.

kernel: 5.7.1
distro: openSUSE Tumbleweed

rfrohl, 2020-06-18:
turns out Dell just uploaded new firmware to lvfs [0], which seems to have fixed the problem. After the update the dock seems to work fine. 

I will reopen the ticket if the problem returns.

For reference I was able to fix the problem with:
```
fwupdmgr refresh --force
fwupdmgr update
```

[0] https://fwupd.org/lvfs/firmware/new

closed 2020-06-18 by rfrohl

oliparcol, 2023-01-20:
I had the same issue with a Dell Inspiron 5431 which was also solved by upgrading the UEFI firmware.