USBGuard/usbguard #392: usbguard 0.7.6 and later crashes on openSUSE 15.2 when started through systemd

lemmy04, 2020-07-07:
openSUSE Leap 15.2 comes with usbguard 0.7.6, and uses the .service file that comes with usbguard... and on Leap 15.2 usbguard-daemon crashes when started through that. One of the many restrictive settings in the .service file might not be compatible with Leap 15.2.

lemmy04, 2020-07-07:
See https://bugzilla.opensuse.org/show_bug.cgi?id=1173750

mgerstner, 2020-07-08:
The problems comes from `SystemCallFilter=@system-service` which seems not to be supported on older systemd versions. Then no system calls at all are allowed and the daemon crashes during execution of the first systemcall with SIGSYS. The warning in the journal is as follows:

```
/usr/lib/systemd/system/usbguard.service:30: Don't know system call group, ignoring: @system-service
```

radosroka, 2020-07-18:
Well, these are packagers issues. They should patch it.

closed 2020-07-18 by radosroka