USBGuard/usbguard #393: use of policykit for access control

lemmy04, 2020-07-08:
I think usbguard should switch to using policykit for access control, instead of using its own rules and configuration files.

Then there would be two sets of rules, one for server systems (only admin users get any access to usbguard) and one for desktop systems (any logged in user has read access to rules and status, and only needs their own user password as authorisation to change / add rules).

Also, bring back a decent (usable) GUI for desktop systems.

muelli, 2020-07-08:
Hm. I'm a bit confused. AFAIK, the access control via DBus is not managed through USBGuard. Right now, I think, you can create PolicyKit rules as you please. Currently, USBGuard offers two IPC mechanisms.

related: https://github.com/USBGuard/usbguard/issues/273 and https://github.com/USBGuard/usbguard/issues/8

Am I missing anything?

radosroka, 2020-07-18:
> I think usbguard should switch to using policykit for access control, instead of using its own rules and configuration files.
>
> Then there would be two sets of rules, one for server systems (only [...]

When you build/install usbguard-dbus you can fully control usbguard via dbus. The default policy kit is included.

hartwork, 2022-02-19:
USBGuard is using PolKit in every sense of the word since PR #531. This ticket can probably be closed.

closed 2022-02-23 by radosroka