USBGuard/usbguard #394: Unable to configure/compile on CentOS 8.2 (pegtl problem)

pipari666, 2020-07-09:
Same error with CentOS 8.2.

Trying to configure with:
./configure --with-crypto-library=sodium --with-bundled-catch --with-bundled-pegtl

Getting error:
....
configure: Using bundled Catch library
checking tao/pegtl.hpp usability... no
checking tao/pegtl.hpp presence... no
checking for tao/pegtl.hpp... no
configure: error: in `/home/username/rpmbuild/BUILD/usbguard-0.7.8':
configure: error: PEGTL header file not found or not usable. Re-run with --with-bundled-pegtl to use the bundled library.

Also tried compiling PEGTL without any success. Official USBGuard release in CentOS8 is not stable. Really need to this.

Please help or point me in any direction!

Best regards

Ilya87, 2020-07-11:
You migh have run git clone https://github.com/USBGuard/usbguard.git and have **no files** in src/ThirdParty/PEGTL, you shoould run git clone --recurse-submodules https://github.com/USBGuard/usbguard.git and then run configure.

Cropi, 2020-08-07:
Did you try to compile it using a [release tarball](https://github.com/USBGuard/usbguard/releases/) or cloned repository?

hartwork, 2022-01-26:
The `PEGTL header file not found or not usable` goes away once you install package `gcc-c++`. But there will be link errors in the end:

```console
# make V=1    # link error in isolation
Makefile:3498: warning: overriding recipe for target 'check'
Makefile:3004: warning: ignoring old recipe for target 'check'
make  all-recursive
Makefile:3498: warning: overriding recipe for target 'check'
Makefile:3004: warning: ignoring old recipe for target 'check'
make[1]: Entering directory '/usbguard'
Making all in src/Tests/
make[2]: Entering directory '/usbguard/src/Tests'
Making all in Fuzzers
make[3]: Entering directory '/usbguard/src/Tests/Fuzzers'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/usbguard/src/Tests/Fuzzers'
make[3]: Entering directory '/usbguard/src/Tests'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/usbguard/src/Tests'
make[2]: Leaving directory '/usbguard/src/Tests'
Makefile:3498: warning: overriding recipe for target 'check'
Makefile:3004: warning: ignoring old recipe for target 'check'
make[2]: Entering directory '/usbguard'
/bin/sh ./libtool  --tag=CXX   --mode=link g++ -pthread -std=c++17   -fvisibility=hidden  -pedantic -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-deprecated-register -Wno-missing-field-initializers -Wno-implicit-fallthrough    -o usbguard src/CLI/usbguard-usbguard.o src/CLI/usbguard-usbguard-list-devices.o src/CLI/usbguard-usbguard-allow-device.o src/CLI/usbguard-usbguard-block-device.o src/CLI/usbguard-usbguard-reject-device.o src/CLI/usbguard-usbguard-apply-device-policy.o src/CLI/usbguard-usbguard-list-rules.o src/CLI/usbguard-usbguard-append-rule.o src/CLI/usbguard-usbguard-remove-rule.o src/CLI/usbguard-usbguard-generate-policy.o src/CLI/usbguard-usbguard-watch.o src/CLI/usbguard-IPCSignalWatcher.o src/CLI/usbguard-PolicyGenerator.o src/CLI/usbguard-usbguard-read-descriptor.o src/CLI/usbguard-usbguard-set-parameter.o src/CLI/usbguard-usbguard-get-parameter.o src/CLI/usbguard-usbguard-add-user.o src/CLI/usbguard-usbguard-remove-user.o ./libusbguard.la  
libtool: link: g++ -pthread -std=c++17 -fvisibility=hidden -pedantic -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-deprecated-register -Wno-missing-field-initializers -Wno-implicit-fallthrough -o .libs/usbguard src/CLI/usbguard-usbguard.o src/CLI/usbguard-usbguard-list-devices.o src/CLI/usbguard-usbguard-allow-device.o src/CLI/usbguard-usbguard-block-device.o src/CLI/usbguard-usbguard-reject-device.o src/CLI/usbguard-usbguard-apply-device-policy.o src/CLI/usbguard-usbguard-list-rules.o src/CLI/usbguard-usbguard-append-rule.o src/CLI/usbguard-usbguard-remove-rule.o src/CLI/usbguard-usbguard-generate-policy.o src/CLI/usbguard-usbguard-watch.o src/CLI/usbguard-IPCSignalWatcher.o src/CLI/usbguard-PolicyGenerator.o src/CLI/usbguard-usbguard-read-descriptor.o src/CLI/usbguard-usbguard-set-parameter.o src/CLI/usbguard-usbguard-get-parameter.o src/CLI/usbguard-usbguard-add-user.o src/CLI/usbguard-usbguard-remove-user.o  ./.libs/libusbguard.so -lqb -lprotobuf -lpthread -lsodium -pthread -Wl,-rpath -Wl,/usr/local/lib
./.libs/libusbguard.so: undefined reference to `vtable for std::filesystem::__cxx11::filesystem_error'
./.libs/libusbguard.so: undefined reference to `typeinfo for std::filesystem::__cxx11::filesystem_error'
./.libs/libusbguard.so: undefined reference to `std::filesystem::__cxx11::filesystem_error::~filesystem_error()'
./.libs/libusbguard.so: undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
./.libs/libusbguard.so: undefined reference to `std::filesystem::__cxx11::filesystem_error::_M_gen_what()'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:1450: usbguard] Error 1
make[2]: Leaving directory '/usbguard'
make[1]: *** [Makefile:2712: all-recursive] Error 1
make[1]: Leaving directory '/usbguard'
make: *** [Makefile:1044: all] Error 2
The command '/bin/sh -c make V=1    # link error in isolation' returned a non-zero code: 2
```

Here's my Dockerfile to see that error in action:
```Dockerfile
# Copyright (c) 2022 Sebastian Pipping <sebastian@pipping.org>
# Licensed under the Apache license version 2.0

FROM centos:8.2.2004
RUN dnf install -y epel-release dnf-plugins-core \
        && \
    dnf config-manager --set-enabled PowerTools \
        && \
    dnf install -y \
        autoconf \
        automake \
        dbus-glib-devel \
        diffutils \
        file \
        gcc-c++ \
        git \
        libqb-devel \
        libgcrypt-devel \
        libsodium-devel \
        libtool \
        libxslt \
        make \
        polkit-devel \
        protobuf-devel \
        protobuf-compiler
RUN git clone --depth 1 https://github.com/USBGuard/usbguard
WORKDIR usbguard
RUN ./autogen.sh
RUN ./configure --with-crypto-library=sodium --with-bundled-catch --with-bundled-pegtl || ! cat config.log
RUN ! make V=1 "-j$(nproc)"  # until the link error
RUN make V=1  # link error in isolation
```

hartwork, 2022-01-26:
PS: A workaround is to use `--without-bundled-pegtl` instead and install package `PEGTL-devel` from EPEL 8. Then it compiles and links without errors for me.

hartwork, 2022-01-27:
PEGTL 3.2.0 in isolation however builds fine:
```Dockerfile
# Copyright (c) 2022 Sebastian Pipping <sebastian@pipping.org>
# Licensed under the Apache license version 2.0

FROM centos:8.2.2004
RUN dnf install -y epel-release dnf-plugins-core \
         && \
    dnf config-manager --set-enabled PowerTools \
         && \
    dnf install -y \
        cmake \
        gcc-c++ \
        git \
        libarchive \
        ninja-build
RUN git clone --depth 1 --branch 3.2.0 https://github.com/taocpp/PEGTL
WORKDIR PEGTL
RUN cmake -B build -G Ninja
RUN ninja -C build -v all
```

I see that PEGTL's build system [is passing `-lstdc++fs` to the linker in some cases](https://github.com/taocpp/PEGTL/blob/7d039707cf835cea63daa78a717e18fcc5bcf95b/CMakeLists.txt#L76), and I have observed GCC 8.5.0 on CentOS 8.2 to be one of these cases:

```console
# docker build .
[..]
Step 5/6 : RUN cmake -B build -G Ninja
 ---> Running in 29ccdf1e9db7
-- The CXX compiler identification is GNU 8.5.0
[..]
-- Adding stdc++fs to the PEGTL to build with std::filesystem.
-- Using std::filesystem
[..]
```

So my understanding is that USBGuard needs to detect when to pass `-lstdc++fs` to the linker for PEGTL.

radosroka, 2022-02-01:
> PS: The ticket could use a rename to show it's true link error nature. Maybe `[master] Link error due to missing -lstdc++fs (e.g. on CentOS 8.2)`?

done.

hartwork, 2022-02-01:
Thank you! :+1:

PR #515 (2022-02-02): Fix compilation for PEGTL 3 with GCC 8 + make GitHub Actions cover GCC 8.5.0 (fixes #394)
Fixes #394

This fixes the [link errors](https://github.com/USBGuard/usbguard/issues/394#issuecomment-1022530220) that happen with PEGTL 3 and GCC 8 on e.g. CentOS 8.2.
The CI part is the same approach as in PR #514 and I plan to rebase one onto the other, once either of them is merged.
I hope you like this direction, I'm looking forward to your review :beers: 

CC @Cropi @radosroka
closed 2022-02-02 by radosroka