USBGuard/usbguard #401: IPC listen gives rules of connected devices in 2 times

LucasParsy, 2020-08-06:
#### current behavior

Inserting an allowed device which has a rule in `rules.conf` fires 2 IPC events:
* one `PresenceChanged` with a `device_rule=block ...`
* immediately after a `PolicyChanged` with `device_rule=allow`

the device is then correctly identified by the system.

#### problem
When using [usbguard-notifier](https://github.com/Cropi/usbguard-notifier), it gives 2 notifications instead of 1, which is confusing and clutters the screen.
For example, when I plug my keyboard/USB hub, it gives me 6 notifications for inserting 3 devices.

It can also be a problem for developer of software listening on the IPC and wanting to execute an action depending on whether a device is allowed or blocked.
(I am working on such a feature (personal initiative), adding an "allow device" action button for inserted blocked device on usbguard-notifier)

<details>
<summary>logs and details:</summary>

tested with usbguard-**0.7.8** (latest release)

rule for my device
```
$ sudo cat /etc/usbguard/rules.conf

...
allow id 13fe:3600 serial "07A702015734A685" name "USB DISK 2.0" hash "YW8O5LdNhI/wN92udBt/HY8AobHXSto/RDQI9g/sryc=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" with-interface 08:06:50 with-connect-type "hotplug"
```

usbguard watch log while inserting my device.
```
$ sudo usbguard watch

[IPC] Connected
[device] PresenceChanged: id=36
 event=Insert
 target=block
 device_rule=block id 13fe:3600 serial "07A702015734A685" name "USB DISK 2.0" hash "YW8O5LdNhI/wN92udBt/HY8AobHXSto/RDQI9g/sryc=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "2-4" with-interface 08:06:50 with-connect-type "hotplug"
[device] PolicyChanged: id=36
 target_old=block
 target_new=allow
 device_rule=allow id 13fe:3600 serial "07A702015734A685" name "USB DISK 2.0" hash "YW8O5LdNhI/wN92udBt/HY8AobHXSto/RDQI9g/sryc=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "2-4" with-interface 08:06:50 with-connect-type "hotplug"
 rule_id=12
```
</details>

#### proposed new behavior:
Inserting an allowed device which has a rule in `rules.conf` fires 1 IPC events:
* a `PresenceChanged` with a `device_rule=allow ...`

The `PolicyChanged` event may be dropped, or kept for compatibility purposes.

<details>
<summary>logs and details:</summary>

example log of usbguard implementing this behavior
```
$ sudo usbguard watch

[IPC] Connected
[device] PresenceChanged: id=36
 event=Insert
 target=block
 device_rule=allow id 13fe:3600 serial "07A702015734A685" name "USB DISK 2.0" hash "YW8O5LdNhI/wN92udBt/HY8AobHXSto/RDQI9g/sryc=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "2-4" with-interface 08:06:50 with-connect-type "hotplug"
```

</details>
I recently discovered USBGuard and I don't know if the underlying code spawning the IPC events, or if there are standards.
So I don't know if my proposition is pertinent or if it can break the flow of existing applications.

<br>
maybe linked to #362

muelli, 2020-08-06:
I claim this is exactly the problem described in https://github.com/USBGuard/usbguard/issues/353 and what https://github.com/USBGuard/usbguard/pull/385 attempts to fix.

LucasParsy, 2020-08-07:
my excuses, you are right, I will close this issue in this case, and follow the other one(s).
(you can maybe delete it?)

closed 2020-08-07 by LucasParsy