USBGuard/usbguard #402: Device is blocked log in console.

Kractucs, 2020-08-07:
Hello guys,

**Configuration**

rules.conf :
`allow with-interface one-of { 03:00:00 03:01:01 03:01:02 }`

**Current behavior**

When a keyboard is plugged in, the devices is authorized but the daemon fires a log in the console :

`usb 1-1: Device is not authorized for usage.`

**Problem**

Even if the device is authorized, the user sees a message in the console. ( -K option for the daemon does not seem to work)

__

**Did I miss configure something ?**

ZoltanFridrich, 2021-02-11:
@Kractucs can you please provide more detail? I am trying to recreate a similar scenario as you have but in fact, usbguard does not even contain string 'for usage' in all of its code, so I am a bit clueless how could you receive log message like:
Device is not authorized for usage.

hartwork, 2022-02-19:
No reply for a long time, should we close?

closed 2022-02-23 by radosroka