USBGuard/usbguard #413: EXCEPTION: stoul

devurandom, 2020-09-04:
I see an error when trying to add a rule for a device that is currently being rejected:
```
> usbguard allow-device -p 'allow id 1050:0112 serial "" name "Yubikey NEO CCID" hash "[REDACTED]" parent-hash "[REDACTED]" with-interface 0b:00:00 with-connect-type "hotplug"'
EXCEPTION: stoul
```

There is no further information, and running the process in `gdb` does not break, so I cannot show you a stack trace.

I am running Gentoo:
```
Portage 3.0.5 (python 3.6.9-final-0, default/linux/amd64/17.1/desktop/plasma/systemd, gcc-10.2.0, glibc-2.32-r1, 5.8.3 x86_64)
=================================================================
                         System Settings
=================================================================
System uname: Linux-5.8.3-x86_64-AMD_Ryzen_5_2400G_with_Radeon_Vega_Graphics-with-gentoo-2.7
KiB Mem:    14194336 total,    724252 free
KiB Swap:          0 total,         0 free
Timestamp of repository gentoo: Wed, 02 Sep 2020 17:45:01 +0000
Head commit of repository gentoo: 6d12b03fb1f0dad081880a78f301297364c376e6
Head commit of repository flatpak-overlay: 26c396aef4d2f8908eea9b3ad2ccf7dc843a0887

Timestamp of repository haskell: Wed, 02 Sep 2020 11:06:42 +0000
Head commit of repository haskell: a1608e03de909304d65d2eab8cacd9ebea1ad347

Head commit of repository local: f5e341d4f6c14425021f6f482ac2f9eb62795d5e

sh bash 5.0_p18
ld GNU gold (Gentoo 2.34 p6 2.34.0) 1.16
ccache version 3.7.11 [disabled]
app-shells/bash:          5.0_p18::gentoo
dev-java/java-config:     2.3.1::gentoo
dev-lang/perl:            5.30.3-r1::gentoo
dev-lang/python:          2.7.18-r1::gentoo, 3.6.12::gentoo, 3.7.9::gentoo, 3.8.5::gentoo, 3.9.0_rc1::gentoo
dev-util/ccache:          3.7.11::gentoo
dev-util/cmake:           3.18.1::gentoo
dev-util/pkgconfig:       0.29.2::gentoo
sys-apps/baselayout:      2.7::gentoo
sys-apps/sandbox:         2.20::gentoo
sys-devel/autoconf:       2.13-r1::gentoo, 2.69-r5::gentoo
sys-devel/automake:       1.13.4-r2::gentoo, 1.16.2::gentoo
sys-devel/binutils:       2.34-r2::gentoo
sys-devel/gcc:            10.2.0-r1::gentoo
sys-devel/gcc-config:     2.3.1::gentoo
sys-devel/libtool:        2.4.6-r6::gentoo
sys-devel/make:           4.3::gentoo
sys-kernel/linux-headers: 5.8::gentoo (virtual/os-headers)
sys-libs/glibc:           2.32-r1::gentoo
Repositories:

gentoo
    location: /var/cache/portage/gentoo
    sync-type: rsync
    sync-uri: rsync://rsync.de.gentoo.org/gentoo-portage
    priority: -1000
    sync-rsync-extra-opts: --new-compress
    sync-rsync-verify-jobs: 1
    sync-rsync-verify-max-age: 24
    sync-rsync-verify-metamanifest: yes

flatpak-overlay
    location: /var/db/repos/flatpak-overlay
    sync-type: git
    sync-uri: https://github.com/fosero/flatpak-overlay.git
    masters: gentoo

haskell
    location: /var/db/repos/haskell
    sync-type: git
    sync-uri: https://github.com/gentoo-mirror/haskell.git
    masters: gentoo
[... 21 lines cut ...]
DISTDIR="/var/cache/portage/distfiles"
EMERGE_DEFAULT_OPTS="--nospinner"
ENV_UNSET="CARGO_HOME DBUS_SESSION_BUS_ADDRESS DISPLAY GOBIN GOPATH PERL5LIB PERL5OPT PERLPREFIX PERL_CORE PERL_MB_OPT PERL_MM_OPT XAUTHORITY XDG_CACHE_HOME XDG_CONFIG_HOME XDG_DATA_HOME XDG_RUNTIME_DIR"
FCFLAGS="-O2 -pipe"
FEATURES="assume-digests binpkg-docompress binpkg-dostrip binpkg-logs buildsyspkg cgroup compressdebug config-protect-if-modified distlocks ebuild-locks fakeroot fixlafiles ipc-sandbox merge-sync mount-sandbox multilib-strict network-sandbox news parallel-fetch parallel-install pid-sandbox preserve-libs protect-owned qa-unresolved-soname-deps sandbox sfperms strict unknown-features-warn unmerge-logs unmerge-orphans userfetch userpriv usersandbox usersync xattr"
FFLAGS="-O2 -pipe"
GENTOO_MIRRORS="http://ftp.spline.inf.fu-berlin.de/mirrors/gentoo/ http://ftp-stud.hs-esslingen.de/pub/Mirrors/gentoo/ http://distfiles.gentoo.org"
LANG="en_US.UTF-8"
LDFLAGS="-Wl,-O1 -Wl,--as-needed -Wl,--hash-style=gnu"
MAKEOPTS="-j6 -l4"
PKGDIR="/var/cache/portage/packages"
PORTAGE_CONFIGROOT="/"
PORTAGE_RSYNC_EXTRA_OPTS="--new-compress"
PORTAGE_RSYNC_OPTS="--recursive --links --safe-links --perms --times --omit-dir-times --compress --force --whole-file --delete --stats --human-readable --timeout=180 --exclude=/distfiles --exclude=/local --exclude=/packages --exclude=/.git"
PORTAGE_TMPDIR="/tmp"
USE="7z 7zip X a52 aac aacplus aacs acl acpi activities aio alsa amd64 appindicator appstream archive audit avahi ayatana bdplus berkdb blake2 blas bluetooth bluray bpf branding brotli bs2b btrfs bzip2 cairo caps cdda cddb cdio cdr celt chromaprint cjk clang cli clipboard color-management colord conntrack crypt cups d3d9 dav1d dbus declarative device-mapper dirac djvu dri drm dts dvb dvd dvdr ed25519 editorconfig egl elf emboss encode epub evdev exif faudio fax fbcon fdk ffmpeg fftw filecaps firefox firewalld fish-completion fits flac fontconfig fontforge fortran fribidi gamepad gbm gdal gdbm geoclue geolocation gif git gles2 gles3 gmp gnome-online-accounts gnupg google googledrive gpg gps graphicsmagick gstreamer gtk gtk3 gui gzip harfbuzz hdf5 heif http2 ibus iconv icu idn imlib inotify introspection ipv6 iwd jemalloc jpeg jpeg2k json kde kipi kms kwallet ladspa lapack latex lcms libatomic libglvnd libidn2 libinput libnotify libproxy libsecret libsoxr libtirpc libvirt lm-sensors lrz lv2 lvm lz4 lzma lzo mad man mariadb markdown mbim mercurial mjpeg mng mobi modemmanager modplug mp3 mp4 mpeg mplayer mpris mtp multilib mysql ncurses netlink networkmanager nftables nls nptl numa office ofono ofx ogg openal opencl opencv openexr opengl openh264 openmax openmp opus pam pango pcap pch pcre pcre2 pdf pgo phonon pixman pkcs11 pkcs7 plasma pm-utils png policykit postscript ppds prison pulseaudio pwquality python qml qrcode qt5 raw rdp readline redfish samba sasl scanner schroedinger screencast sctp sdl sdl2 seccomp semantic-desktop share smartcard snappy sparse speech speex spell spice ssl startup-notification steamruntime stemmer svg systemd systemtap tbb tcpd teamd telemetry telepathy tga theora threads thunderbolt tiff timezone tmux truetype tslib udev udisks uinput unicode unwind upnp upnp-av upower usb utempter v4l v4l2 vaapi vdpau vkd3d vorbis vpx vulkan wasm wavpack wayland webchannel webengine webp widgets wireguard wmf woff2 wps x264 x265 xattr xcb xinerama xkb xml xmp xrandr xscreensaver xv xvid xwayland xxhash xz yaml zeroconf zeromq zimg zlib zstd" ABI_X86="64" ADA_TARGET="gnat_2018" ALSA_CARDS="hda-intel" APACHE2_MODULES="authn_core authz_core socache_shmcb unixd actions alias auth_basic authn_alias authn_anon authn_dbm authn_default authn_file authz_dbm authz_default authz_groupfile authz_host authz_owner authz_user autoindex cache cgi cgid dav dav_fs dav_lock deflate dir disk_cache env expires ext_filter file_cache filter headers include info log_config logio mem_cache mime mime_magic negotiation rewrite setenvif speling status unique_id userdir usertrack vhost_alias" CALLIGRA_FEATURES="karbon sheets words" CAMERAS="ptp2" COLLECTD_PLUGINS="df interface irq load memory rrdtool swap syslog" CPU_FLAGS_X86="aes avx avx2 f16c fma3 mmx mmxext pclmul popcnt rdrand sha sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3" ELIBC="glibc" ENLIGHTENMENT_MODULES="*" GPSD_PROTOCOLS="ashtech aivdm earthmate evermore fv18 garmin garmintxt gpsclock greis isync itrax mtk3301 nmea ntrip navcom oceanserver oldstyle oncore rtcm104v2 rtcm104v3 sirf skytraq superstar2 timing tsip tripmate tnt ublox ubx" GRUB_PLATFORMS="efi-64" INPUT_DEVICES="joystick libinput" KERNEL="linux" L10N="de de-DE en en-GB ar fa tr ja ko zh zh-CN zh-TW" LCD_DEVICES="bayrad cfontz cfontz633 glk hd44780 lb216 lcdm001 mtxorb ncurses text" LIBREOFFICE_EXTENSIONS="nlpsolver scripting-javascript wiki-publisher" LIRC_DEVICES="devinput" LLVM_TARGETS="AMDGPU BPF RISCV WebAssembly" LUA_TARGET="lua5-2" OFFICE_IMPLEMENTATION="libreoffice" PHP_TARGETS="php7-2 php7-3" POSTGRES_TARGETS="postgres10 postgres11" PYTHON_SINGLE_TARGET="python3_7" PYTHON_TARGETS="python2_7 python3_7 pypy pypy3" QEMU_SOFTMMU_TARGETS="riscv32 riscv64 x86_64" QEMU_USER_TARGETS="riscv32 riscv64" RUBY_TARGETS="ruby25" USERLAND="GNU" VIDEO_CARDS="amdgpu virgl" XTABLES_ADDONS="quota2 psd pknock lscan length2 ipv4options ipset ipp2p iface geoip fuzzy condition tee tarpit sysrq steal rawnat logmark ipmark dhcpmac delude chaos account"
Unset:  CC, CPPFLAGS, CTARGET, CXX, INSTALL_MASK, LC_ALL, LINGUAS, PORTAGE_BINHOST, PORTAGE_BUNZIP2_COMMAND, PORTAGE_COMPRESS, PORTAGE_COMPRESS_FLAGS

=================================================================
                        Package Settings
=================================================================

sys-apps/usbguard-0.7.8::gentoo was built with the following:
USE="dbus (policykit) systemd -bash-completion -ldap -static-libs" ABI_X86="(64)"
```

Cropi, 2020-09-04:
Hi,

I think that the rule should not be a single a argument(omit the apostrophes and escape the quotes)
Maybe something like this:
```
> usbguard allow-device -p allow id 1050:0112 serial \"\" name \"Yubikey NEO CCID\" hash \"\[REDACTED\]\" parent-hash \"\[REDACTED\]\" with-interface 0b:00:00 with-connect-type \"hotplug\"
```

Anyway, I think it would be better to use it as you described.

devurandom, 2020-09-04:
I also tried that, but `usbguard` responded with an also-not-very-descriptive `ERROR: RuleParserError` and the usage information.

In any case, I think the error messages could be improve a bit.

ZoltanFridrich, 2020-09-04:
EXCEPTION: stoul happens when you give usbguard allow-device a single argument. USBGuard tries to parse that argument as an ID of a device and that fails.

radosroka, 2020-09-28:
Do you have still problem with that?

devurandom, 2020-09-28:
Was anything done to improve usbguard's behaviour in the cases explained above, especially with regards to error messages?

PR #424 (2020-10-19): Extend allow/block/reject-device commands to accept a rule as a single argument(in apostrophes)
Commit ecd99dba7a94708e9a4a72ca831bde283db3ec26 ensures that ```allow-device``` command will accept a rule as a single argument, so we do not have to escape special characters. Old behavior can also be used.
Besides that this commit fixes an error, when the rule consists of only one argument, e.g. ```usbguard allow-device block``` for allowing all blocked devices (fixes #413).
The same applies for ```block-device``` and ```reject-device```.

Commit 534236680efc80ec508e801ed3c27f75d8ee339f fixes two  [... cut]
closed 2020-10-19 by radosroka