USBGuard/usbguard #415: usbguard on android

carl1314, 2020-09-08:
I currently urgently need to implement the usb firewall function on android, so I plan to bring up usbguard to android.Could you kindly give some guide or suggestion ?

radosroka, 2020-09-28:
Hello,

I've no experience with writing, building or deploying of the USBGuard...
I believe that it is possible.

genodeftest, 2020-09-30:
Duplicate of #414.

SkewedZeppelin, 2020-10-30:
GrapheneOS and LineageOS (and their derivatives) support disallowing USB devices when locked.
I've linked the corresponding patches below for reference.

Kernel
Only use the one specific to your kernel version.
GPLv2
https://github.com/Divested-Mobile/kernel_patches/blob/master/0005-Copperhead-Deny_USB/3.4/3.4-Backport.patch
https://github.com/Divested-Mobile/kernel_patches/blob/master/0005-Copperhead-Deny_USB/3.10/0001.patch
https://github.com/Divested-Mobile/kernel_patches/blob/master/0005-Copperhead-Deny_USB/3.10/0001-Alt.patch
https://github.com/Divested-Mobile/kernel_patches/blob/master/0005-Copperhead-Deny_USB/3.18/0002.patch
https://github.com/Divested-Mobile/kernel_patches/blob/master/0005-Copperhead-Deny_USB/3.18/0002-Alt1.patch
https://github.com/Divested-Mobile/kernel_patches/blob/master/0005-Copperhead-Deny_USB/3.18/0002-Alt2.patch
https://github.com/Divested-Mobile/kernel_patches/blob/master/0005-Graphene-Deny_USB/4.4/0001.patch
https://github.com/Divested-Mobile/kernel_patches/blob/master/0005-Graphene-Deny_USB/4.9/0002.patch
https://github.com/Divested-Mobile/kernel_patches/blob/master/0005-Graphene-Deny_USB/4.14/0003.patch

GrapheneOS aosp/11.0
Apache-2.0
https://github.com/GrapheneOS/platform_build/commit/cef798631c804b59f516f863b69b884c76f064b1
https://github.com/GrapheneOS/platform_frameworks_base/commit/f193967333576ec90b3ae10a995dd279c60e2dd1
https://github.com/GrapheneOS/platform_packages_apps_Settings/commit/459bb88ca7bbafe4eea8ac7bf3e4cb3e7f7bddd7
https://github.com/GrapheneOS/platform_system_core/commit/3c0b76cba4ce563634f8fc777e7c0938f83e435b
https://github.com/GrapheneOS/platform_system_sepolicy/commit/a3f8c4dba18deeab9628d563cbdf326d11328056

GrapheneOS aosp/10.0
Apache-2.0
https://github.com/GrapheneOS/platform_build/commit/30f293b110a77c4237fbc1b0a67f54ade3565053
https://github.com/GrapheneOS/platform_frameworks_base/commit/f4acb713674e1d0b860888264430431f97e21077
https://github.com/GrapheneOS/platform_packages_apps_Settings/commit/dd79b1ee87a10d7c8cba91207bc12a1932ad43a5
https://github.com/GrapheneOS/platform_system_core/commit/2a7b688f52e04c0462e4058fcb1d85b51b18a119
https://github.com/GrapheneOS/platform_system_sepolicy/commit/73c1e104d523b6e28528e9f17416cf784d9c9e97

GrapheneOS aosp/9.0
Apache-2.0
https://github.com/GrapheneOS/platform_build/commit/d43bde5d0019f650ddc01e9ada2f2d04b19b6ae1
https://github.com/GrapheneOS/platform_frameworks_base/commit/8788c25eadf803c38e28f3c61a602911dbe9fdee
no settings patch available
https://github.com/GrapheneOS/platform_system_core/commit/d56c8718441c56167488e6522e1a0756bf738f87
https://github.com/GrapheneOS/platform_system_sepolicy/commit/6ddd694acd98515bb96378d2379d73f02af846ae

LineageOS 18.0/11.0
Apache-2.0
https://review.lineageos.org/c/LineageOS/android_frameworks_base/+/287519
along with the others from https://review.lineageos.org/q/topic:%22trust-usb%22

LineageOS 17.1/10.0
Apache-2.0
https://review.lineageos.org/c/LineageOS/android_frameworks_base/+/258803
along with the others from https://review.lineageos.org/q/topic:%22trust-usb%22

LineageOS 16.0/9.0
Apache-2.0
https://review.lineageos.org/q/topic:%22trust-usb%22

LineageOS 15.1/8.1
These following patches are CC BY-NC-SA.
That means NO commercial use!
You must provide credit for their use!
https://creativecommons.org/licenses/by-nc-sa/3.0/
https://github.com/Divested-Mobile/DivestOS-Build/blob/39740b384f2ae44d1663641a207f03c0dbd92a9b/Patches/LineageOS-15.1/android_build/Copperhead/0002-Deny_USB.patch
https://github.com/Divested-Mobile/DivestOS-Build/blob/39740b384f2ae44d1663641a207f03c0dbd92a9b/Patches/LineageOS-15.1/android_frameworks_base/Copperhead/0005-Deny_USB.patch
https://github.com/Divested-Mobile/DivestOS-Build/blob/39740b384f2ae44d1663641a207f03c0dbd92a9b/Patches/LineageOS-15.1/android_packages_apps_Settings/Copperhead/0003-Deny_USB.patch
https://github.com/Divested-Mobile/DivestOS-Build/blob/39740b384f2ae44d1663641a207f03c0dbd92a9b/Patches/LineageOS-15.1/android_system_core/Copperhead/0002-Deny_USB.patch
https://github.com/Divested-Mobile/DivestOS-Build/blob/39740b384f2ae44d1663641a207f03c0dbd92a9b/Patches/LineageOS-15.1/android_system_sepolicy/Copperhead/0002-Deny_USB.patch