USBGuard/usbguard #420: USB4 / Thunderbolt support in usbguard?

rajatxjain, 2020-09-22:
Hello,

The USB4 and Thunderbolt spec allows for a alternate mode that requires the thunderbolt's "authorized" sysfs bit also to be  toggled to enable and authorize the USB devices plugged into either a USB4/thunderbolt dock or directly to a thunderbolt port:

Essentially:

echo 1 > /sys/bus/thunderbolt/<device>/authorized

would enable / authorize the USB devices plugged downstream.

We (Google) would like to expand the code of usbguard to rake care of thunderbolt/USB4 devices and implementing whitelist/blacklist of USB4/thunderbolt devices. Do you think it makes sense? Can you provide inputs to how big of a change would this be, and possibly provide pointers to code that would need change?

Thanks,

Rajat

gicmo, 2020-09-23:
Thunderbolt devices authorization (devices in `/sys/bus/thunderbolt` of type `thunderbolt_device`) are already handled by [bolt](https://github.com/gicmo/bolt), which does already have individual device policy management and is integrated into GNOME / KDE (and is shipped by default in all major distros). This will also work for USB4 / Thunderbolt.
You can have a look there to see how much code that would be, if you wanted to have the same logic in usbguard. NB: for modern systems with IOMMU the general policy is to just authorize them, but there is a global flag in bolt to disable that. What exactly are you after? A uuid based allow / deny-list?

rajatxjain, 2020-09-24:
Yes, we're looking to implement UUID based allowlist. I'm looking at bolt in parallel but it seems like it has a lot of dependencies on polkit (something we don't ship currently). We already use usb_guard today for USB devices, and were hoping it would be easy to extend that for thunderbolt devices, but when I started looking at code, it seemed to me that there are a lot of assumptions about  the it being a USB device (sysfs attributes, paths etc). 

Trying to gauge usefulness of this effort though. Do you think in general this is something that would be useful / interesting outside of our usecase? (Or would going through bolt make more sense)?

gicmo, 2020-09-24:
> Yes, we're looking to implement UUID based allowlist. I'm looking at bolt in parallel but it seems like it has a lot of dependencies on polkit (something we don't ship currently).

Polkit integration is only used for authenticating the D-Bus calls and is very neatly separated via `BoltBouncer` ([`bolt-bouncer.c`](https://gitlab.freedesktop.org/bolt/bolt/-/blob/master/boltd/bolt-bouncer.c)). Replacing that with something else and making the dependency on Polkit optional should not be a huge amount of work. I am happy to help out here. 
Are you shipping D-Bus? bolt is more tightly depended on D-Bus itself, although that is also very much abstracted away to a single class as well (`BoltExported`). But replacing that would be more work, but could also be done. If that is more or less then re-purposing USBGuard for Thunderbolt is hard for me to say because I don't know the USBGuard code base that well. 

> Trying to gauge usefulness of this effort though. Do you think in general this is something that would be useful / interesting outside of our usecase? (Or would going through bolt make more sense)?

I let the USBGuard maintainer also chime in here, maybe just two things: 1) I am (as bolt maintainer) happy to take patches and also help out myself to make it fit more your usecase, 2) if Thunderbolt support was added to USBGuard, it would either be an optional feature to not interfere with bolt on normal Linux installations or complete (boot acl, iommu, D-Bus API) to basically replace bolt in the future. The latter, I think, would be quite a lot of work.

rajatxjain, 2020-09-24:
> Polkit integration is only used for authenticating the D-Bus calls and is very neatly separated via BoltBouncer (bolt-bouncer.c). Replacing that with something else and making the dependency on Polkit optional should not be a huge amount of work. I am happy to help out here.

This is great! This is for integration into Chromiumos / Chromeos BTW. I tried importing bolt into our gentoo build and got sucked into importing successive dependencies (polkit, spidermonkey, umockdev...) and eclasses into our build, and eventually gave up because of polkit issues mostly. If you can please help by providing a patch that removes the polkit dependency altogether (may be start with unauthenticated dbus, if that is a thing?), that would really help us.  

> Are you shipping D-Bus? bolt is more tightly depended on D-Bus itself, although that is also very much abstracted away to a single class as well (BoltExported). But replacing that would be more work, but could also be done. 

Yes, we're shipping dbus. 

> I let the USBGuard maintainer also chime in here, maybe just two things: 1) I am (as bolt maintainer) happy to take patches and also help out myself to make it fit more your usecase, 2) if [...]

Our goal is not to replace bolt, but the primary motivation of exploring usbguard was that it was already in use in our distro. I think it makes sense to import bolt instead since it seems like it does support thunderbolt already. Is there a mailing list for bolt (or discussion on it also via the github "issues" like this?)?

Thank you so much!

gicmo, 2020-09-25:
> This is great! This is for integration into Chromiumos / Chromeos BTW. I tried importing bolt into our gentoo build and got sucked into importing successive dependencies (polkit, spidermonkey, [...]

Yeah, I will look into this. Should not be too hard. `umockdev` is "only" needed for the test framework.

> Our goal is not to replace bolt, but the primary motivation of exploring usbguard was that it was already in use in our distro. I think it makes sense to import bolt instead since it seems like it [...]

Issues over at https://gitlab.freedesktop.org/bolt/bolt makes the most sense.

radosroka, 2020-09-28:
Hello,

I was asked about this support multiple times in the past. I feel like this does not fit into the USBGuard concept.
I like the simplicity of current solution that just works. I believe it is much better (for security) to have (in this case two) simple daemons which are sandboxed by systemd, apparmor/selinux.. My concern is about the USBGuard getting "capabilities" to interfere with PCI and other stuff you can do with thunderbolt. There is not much to do for sandboxed USBGuard and that is good. Boltd is also much more secure with less code => simplicity. Another problem could be exponentially bigger attack vector. For example USB security can be affected(turned off) by some bug with thunderbolt and vice versa.

I think that we would need to redesign and rewrite most of the project. The complexity of the future solution would increase exponentially... It would be very likely error prone and buggy.

I would rather see the this UUID whitelist implemented in boltd. But if there will be some strong justification about opposite I can still reconsider that. If so I would like to see some high level proposal and design change.

gicmo, 2020-09-28:
I have had a quick go at the bolt-no-polkit in [MR#222](https://gitlab.freedesktop.org/bolt/bolt/-/merge_requests/222). That should already work. I guess you would use the standard d-bus policy to configure the policies more (I dont know how D-Bus is used in Chromiumos / Chromeos)? If so, we could ship that in bolt and use that in case of no-polkit. 

I am also happy to add some more code to have a UUID allow-list, if that is needed.

rajatxjain, 2020-10-29:
Thank you Christian for MR#222! 

This is just to let you know that I got consumed temporarily with an alternate urgent task, but I will get back to this and let you know how this went. 

[I ran into some gentoo build issues when I was compiling the bolt from MR#222 from within our gentoo build subsystem. It likely doesn't have anything to do with bolt though, so I will take a look at it, after my current task] 

Thank you again for your help and support, and I do intend to use the MR#222. I have 1 question though:

> I am also happy to add some more code to have a UUID allow-list, if that is needed.

Oh. I though this (maintaining a UUID based database of allowed devices) is something already present in bolt? 

Thanks,

Rajat

radosroka, 2022-02-23:
The issue is not relevant to usbguard anymore.

closed 2022-02-23 by radosroka