USBGuard/usbguard #423: musl libc compatibility

mpolanski, 2020-10-11:
I tried to use USBGuard on Alpine Linux, but I doesn't work on musl-based distributions. I successfully built it using a patch provided by Void Linux: https://github.com/void-linux/void-packages/blob/427d50c6a66e4a1338dbe230a41b98a2419116a9/srcpkgs/usbguard/patches/musl.patch, but it's not enough. At startup USBGuard tries to parse /etc/nsswitch.conf file, which is present only on glibc systems.

radosroka, 2020-10-11:
I believe you can add this file. It just needs to be there even empty... You can install nsswitch.conf by usbguard package.

radosroka, 2020-10-11:
It should be easy to patch usbguard so the file won't be necessary.

radosroka, 2020-10-11:
Try this patch: https://github.com/USBGuard/usbguard/commit/b1b1f4bed9507d3c547d174f67be9e3a9c2e5d99

mpolanski, 2020-10-11:
Thanks for preparing the patch. Works as expected.

closed 2020-10-11 by radosroka

mpolanski, 2020-10-18:
USBGuard is now available on Alpine Linux (in the testing repository): https://pkgs.alpinelinux.org/packages?name=usbguard&branch=edge

PR #525 (2022-02-14): daemon: Be okay without glibc file /etc/nsswitch.conf e.g. with musl on Alpine (fixes #423)
Fixes #423
Related #524

Note: LGTM failure is unrelated, see #528