USBGuard/usbguard #432: [0.7.8] ./.libs/libusbguard.so: error: undefined reference to 'pthread_cond_clockwait'

hartwork, 2020-11-24:
Hi!

I am aware that USBguard does not use `pthread_cond_clockwait` directly, yet thee is a [downstream bug report](https://bugs.gentoo.org/754906) with a link error like this:
```
./configure --prefix=/usr --build=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --mandir=/usr/share/man --infodir=/usr/share/info --datadir=/usr/share --sysconfdir=/etc --localstatedir=/var/lib --disable-dependency-tracking --disable-silent-rules --docdir=/usr/share/doc/usbguard-0.7.8 --htmldir=/usr/share/doc/usbguard-0.7.8/html --with-sysroot=/ --libdir=/usr/lib64 --with-dbus --without-ldap --with-polkit --disable-static --enable-systemd
[..]
      libqb: system-wide; -pthread -lqb
     crypto: system-wide;  -lsodium
   libaudit: system-wide; -laudit
    libldap: None; building without LDAP support
 libseccomp: system-wide; -lseccomp
  libcap-ng: system-wide; -lcap-ng
   protobuf: system-wide; -pthread -lprotobuf
      Catch: system-wide; -I/usr/include/catch 
      PEGTL: system-wide;  ; version <= 2.6.0: 
      GDBus: system-wide; -pthread -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/usr/include/libmount -I/usr/include/blkid -I/usr/lib64/libffi/include -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -ldbus-1 -lgio-2.0 -lgobject-2.0 -lglib-2.0
   umockdev: system-wide; -lumockdev -lgobject-2.0 -lglib-2.0
[..]
libtool: link: x86_64-pc-linux-gnu-g++ -std=c++11 -pipe -O2 -march=znver1 -fvisibility=hidden -pedantic -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-deprecated-register -Wno-missing-field-initializers -Wno-implicit-fallthrough -Wl,-O1 -Wl,--as-needed -Wl,--hash-style=gnu -o .libs/usbguard-rule-parser src/CLI/usbguard_rule_parser-usbguard-rule-parser.o  ./.libs/libusbguard.so -lqb -lprotobuf -lsodium -lumockdev -lgobject-2.0 -lglib-2.0
./.libs/libusbguard.so: error: undefined reference to 'pthread_cond_clockwait'
./.libs/libusbguard.so: error: undefined reference to 'pthread_create'
[..]
```

I have not been able to reproduce it yet. This is with GCC 10.2.0 but probably not specific to that version.
Any ideas how to best proceed?

devurandom, 2021-01-16:
https://github.com/USBGuard/usbguard/pull/445 should fix this. Can you please try, @hartwork?

hartwork, 2021-01-16:
@devurandom hi! I could not reproduce the initial report on my own system locally so my feedback on the proposed fixed version is of limited value: If it links fine, there is no new information. Does that make sense? Any ideas how to enable me to help for real?

closed 2021-01-18 by radosroka