USBGuard/usbguard #433: Unable to permanently allow device

ibahnasy, 2020-11-24:
When I try to permanently allow a device I get this error:
IPC ERROR: request id=1: FileRuleSet saving: /etc/usbguard/rules.conf: Read-only file system

distro: Arch
usbguard 0.7.5

# cat /usr/lib/systemd/system/usbguard.service 
[Unit]
Description=USBGuard daemon
Wants=systemd-udevd.service local-fs.target
Documentation=man:usbguard-daemon(8)

[Service]
AmbientCapabilities=
CapabilityBoundingSet=CAP_CHOWN CAP_FOWNER
DeviceAllow=/dev/null rw
DevicePolicy=strict
ExecStart=/usr/bin/usbguard-daemon -k -c /etc/usbguard/usbguard-daemon.conf
IPAddressDeny=any
LockPersonality=yes
MemoryDenyWriteExecute=yes
NoNewPrivileges=yes
PrivateDevices=yes
PrivateTmp=yes
ProtectControlGroups=yes
ProtectHome=yes
ProtectKernelModules=yes
ProtectSystem=yes
PIDFile=/run/usbguard.pid
ReadOnlyPaths=-/
ReadWritePaths=-/dev/shm -/var/log/usbguard -/tmp
Restart=on-failure
RestrictAddressFamilies=AF_UNIX AF_NETLINK
RestrictNamespaces=yes
RestrictRealtime=yes
SystemCallArchitectures=native
SystemCallFilter=@system-service
Type=simple
UMask=0077

[Install]
WantedBy=basic.target

Cropi, 2020-11-25:
The 0.7.5 version was released on 28.06.2019. Since that version there were a couple of notable changes. I think that upgrading to version 0.7.8 would solve your problem.
Anyway, what kind of permission/ownership does your /etc/usbguard/rules.conf have?

ibahnasy, 2020-11-25:
-rw------- 1 root root 4274 Apr  1  2020 /etc/usbguard/rules.conf

Cropi, 2020-11-25:
You might need to add ```-/etc/usbguard/``` to ReadWritePaths in your service file.

hartwork, 2022-01-28:
I see that [Arch has USBGuard 1.0.0 by now](https://archlinux.org/packages/extra/x86_64/usbguard/) and the `usbguard.service` file installed contains an entry for `/etc/` in key `ReadWritePaths` now, the `-%sysconfdir%/usbguard/` one here:
https://github.com/USBGuard/usbguard/blob/b9ea78e195941423c7b797624fc139a0e3c38c66/usbguard.service.in#L23

@ibahnasy does the issue still occur or can this be closed as fixed?

hartwork, 2022-02-19:
My vote for closing. Seems obsolete, needs a reply, no reply for 3 weeks.

closed 2022-02-23 by radosroka