USBGuard/usbguard #435: Segmentation fault / doubled freed after allow

monolidth, 2020-11-26:
Hey, 
I'm currently using **arch linux** (`uname -r` : _5.9.9-arch1-1_).  If I try to allow some devices, like an mouse or and
usb stick, sometime USBGUARD seg fault. It is hard to reproduce since sometime usbguard does not whitelist (without any error).

## Services
[image]

## Steps to redproduce
1. Show devices: `usbguard list-devices`
```
... 
21: block id 1ea7:1006 serial "" name "Gaming Mouse" hash "Qr+2v09hJbOq5jfrSjXaW8rgGhKx7kRuHDwWF+L92v0=" parent-hash "4a4NgfdUaJO43rkCzmWRSeHHR/uUh5+SNsXnhosm9qs=" via-port "2-2" with-interface { 03:01:02 03:00:01 } with-connect-type "hotplug"
```

2. Allow new device with `usbguard allow-device 21`
Return value is zero but device is still blocked. Sometime USBGUARD crashes as descripted.

## Edit: (doubled freed)
[image]

I strongly recommend to use an issue template and I'm open to any solutions.
Regards,

William

Cropi, 2020-12-02:
There was a very simmilar issue to this one in the past (https://github.com/USBGuard/usbguard/issues/329) but we could not reproduce it, so we thought it was not a problem anymore.

I did some digging and I was finally able to reproduce this issue. After spending some time on debugging I can confirm that ```IPCClientPrivate::disconnect(...)``` is not thread safe (https://github.com/USBGuard/usbguard/blob/master/src/Library/IPCClientPrivate.cpp#L139).  There is a small chance that two threads would deallocate the same ipc resources, will try to fix it.

monolidth, 2020-12-04:
I compiled this new branch and it seems to work. Thank you for your fast respond.

PR #437 (2020-12-09): Double free fix
Commit https://github.com/USBGuard/usbguard/commit/93088a9107b77e588816913b825999825ab8e86d makes IPCClientPrivate@disconnect thread safe, so no double free happens after deallocating IPC resources, fixes #435
commit 77b70997 (2020-12-09): Make IPCClientPrivate@disconnect thread safe
  This commit prevents a sefault when 2 or more threads would try to
  deallocate sources
  fixes #435
closed 2020-12-08 by monolidth