USBGuard/usbguard #436: Feature resquest: Temporarily detect and allow new devices

FedericoCeratto, 2020-12-02:
It would be very convenient if usbguard could be instructed to allow any new device plugged in **after** the command.
Optionally with a time window e.g. 10 seconds.
Also with an optional flag to create a permanent rule.

Cropi, 2020-12-02:
> It would be very convenient if usbguard could be instructed to allow any new device plugged in **after** the command.

What do you mean by "after the command"?
There is a very similar feature that we are working on: https://github.com/Cropi/usbguard-notifier/pull/62.
> Optionally with a time window e.g. 10 seconds.
> Also with an optional flag to create a permanent rule.

By default, all rules are permanent. If you want to change the policy of a device permanently, you might want to use the CLI, like:
```usbguard allow-device -p device_id``` , where **-p** means permanent.

FedericoCeratto, 2020-12-02:
> What do you mean by "after the command"?

1. The user runs something like `usbguard allow-new-devices`
1. The user plugs in one or more new devices (never seen before) and they are all allowed

This can be useful when user are in a trusted environment, and especially useful when using application that cannot handle a USB device being blocked even for a second.

> By default, all rules are permanent.

According to the manpage, allow-device requires "-p" to make it permanent, implying that the default behavior is temporary.

Cropi, 2020-12-02:
> > What do you mean by "after the command"?
> 
>     1. The user runs something like `usbguard allow-new-devices`
> 
>     2. The user plugs in one or more new devices (never seen before) and they are all allowed
> 
> 
> This can be useful when user are in a trusted environment, and especially useful when using application that cannot handle a USB device being blocked even for a second.
> 
> > By default, all rules are permanent.
> 
> According to the manpage, allow-device requires "-p" to make it permanent, implying that the default behavior is temporary.

Sorry, I didn't mean that. When you mentioned "create permanent rules" I thought you were using the ```usbguard append-rule``` CLI command and that's why I wrote rules are permanent by default.

ZoltanFridrich, 2021-02-18:
There is a similar feature currently in development for usbguard-notifier. The feature will enable the user to allow devices directly through notifications.

genodeftest, 2021-06-03:
@FedericoCeratto wrote:
> It would be very convenient if usbguard could be instructed to allow any new device plugged in after the command.
Optionally with a time window e.g. 10 seconds.

This could be done if you wrote a shell script or alias which does something like this:
```
usbguard set-parameter ImplicitPolicyTarget allow
sleep 10
usbguard set-parameter ImplicitPolicyTarget block
```

This minimal solution does not however fulfil your second requirement:
>  Also with an optional flag to create a permanent rule.