USBGuard/usbguard #441: USBGuard doesn't compile with pegtl 3.0

xelaseer, 2020-12-12:
As per the title. 

USBGuard 0.7.8 fails with the following errors while building:
```
checking tao/pegtl.hpp presence... yes
configure: WARNING: tao/pegtl.hpp: present but cannot be compiled
configure: WARNING: tao/pegtl.hpp:     check for missing prerequisite headers?
configure: WARNING: tao/pegtl.hpp: see the Autoconf documentation
configure: WARNING: tao/pegtl.hpp:     section "Present But Cannot Be Compiled"
configure: WARNING: tao/pegtl.hpp: proceeding with the compiler's result
configure: WARNING:     ## ------------------------------------ ##
configure: WARNING:     ## Report this to usbguard@usbguard.org ##
configure: WARNING:     ## ------------------------------------ ##
checking for tao/pegtl.hpp... no
configure: error: in `/var/tmp/portage/sys-apps/usbguard-0.7.8/work/usbguard-0.7.8':
configure: error: PEGTL header file not found or not usable. Re-run with --with-bundled-pegtl to use the bundled library.
```
Pegtl 3.0 fixes some errors with -werror-type-limits, which were causing build failures on my computer with 2.8.3, so USBGuard support for the newer Pegtl 3.0 would be much appreciated.

b1rger, 2020-12-31:
Hi, I did some research and there are a couple of changes that have to be made for usbguard to work with pegt 3.x. The first thing to do is to bump the supported C++ standard to C++17 in `configure.ac`. Then the file `tao/pegtl/contrib/tracer.hpp` has been dropped and replaced by `tao/pegtl/contrib/trace.hpp` - this affects `src/Library/UEventParser.cpp`, `src/Library/UMockdevDeviceDefinition.cpp`, `src/Library/public/usbguard/RuleParser.cpp`, `src/Tests/Fuzzers/fuzzer-rules.cpp` and `src/Tests/Fuzzers/fuzzer-uevent.cpp`. Following that the call to `tao::pegtl::parse` in those files has to be updated, but thats where I'm stuck - I don't have enough C++ skills to know how to update that :confused:

hartwork, 2021-01-31:
This is [causing trouble](https://bugs.gentoo.org/767469) in Gentoo with pegtl 3.2.0 as well. I'd want to not use the bundled pegtl [for security reasons](https://wiki.gentoo.org/wiki/Why_not_bundle_dependencies#Security_implications) and keeping users in the past. So please take my vote for supporting pegtl 3.2.0. Many thanks! :pray:

PR #453 (2021-03-01): Add support for PEGTL 3.2.0
No configuration switches need to be implemented. Now, USBGuard seems to be working with both new and old versions of PEGTL.

Resolves #441
closed 2021-03-01 by Cropi