USBGuard/usbguard #442: USBGuard 0.7.8 failed to start using Systemd on Ubuntu 18.04

aditiambadkar, 2020-12-15:
I have installed USBGuard 0.7.8 on Ubuntu 18.04 using the following steps:

```
aditi@aditi-VirtualBox:~$ git clone https://github.com/USBGuard/usbguard.git

aditi@aditi-VirtualBox:~$ cd usbguard/

aditi@aditi-VirtualBox:~/usbguard$ git checkout usbguard-0.7.8 

aditi@aditi-VirtualBox:~/usbguard$ sudo apt install dh-autoreconf pkg-config

aditi@aditi-VirtualBox:~/usbguard$ autoreconf -i -f

aditi@aditi-VirtualBox:~/usbguard$ sudo apt install libqb-dev libtool libsodium-dev libprotobuf-dev protobuf-compiler libdbus-glib-1-dev libxml2-utils xsltproc libpolkit-gobject-1-dev libgcrypt20-dev libssl-dev libcrypto++6 libcrypto++-dev libcrypto++6-dbg asciidoc asciidoctor libseccomp-dev seccomp libcap-ng-dev

aditi@aditi-VirtualBox:~/usbguard$ ./autogen.sh

aditi@aditi-VirtualBox:~/usbguard$ ./configure --prefix=/usr --sysconfdir=/etc --with-bundled-catch --with-bundled-pegtl --with-crypto-library=sodium --enable-systemd

aditi@aditi-VirtualBox:~/usbguard$ make

aditi@aditi-VirtualBox:~/usbguard$ sudo make install

aditi@aditi-VirtualBox:~/usbguard$ make check

aditi@aditi-VirtualBox:~/usbguard$ sudo ldconfig
 
aditi@aditi-VirtualBox:$ sudo usbguard generate-policy > rules.conf

aditi@aditi-VirtualBox:~$ sudo install -m 0600 -o root -g root rules.conf /etc/usbguard/rules.conf

aditi@aditi-VirtualBox:~$ sudo systemctl daemon-reload

aditi@aditi-VirtualBox:~$ sudo systemctl start usbguard.service

aditi@aditi-VirtualBox:~$ sudo systemctl enable usbguard.service

aditi@aditi-VirtualBox:~$ sudo systemctl status usbguard.service
```

This is the output of ```sudo systemctl status usbguard.service```

```
aditi@aditi-VirtualBox:~$ sudo systemctl status usbguard.service 
● usbguard.service - USBGuard daemon
   Loaded: loaded (/lib/systemd/system/usbguard.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-12-15 13:54:19 +04; 23min ago
     Docs: man:usbguard-daemon(8)
 Main PID: 21576 (usbguard-daemon)
    Tasks: 3 (limit: 4915)
   CGroup: /system.slice/usbguard.service
           └─21576 /usr/sbin/usbguard-daemon -k -c /etc/usbguard/usbguard-daemon.conf

Dec 15 13:54:19 aditi-VirtualBox systemd[1]: Started USBGuard daemon.
Dec 15 13:54:19 aditi-VirtualBox usbguard-daemon[21576]: [1608026059.927] (A) uid=0 pid=21576 device.rule='allow id 1d6b:0002 serial "0000:00:0b.0" name "EHCI Host Controller" hash "SEiVqUWwefEKDMN9OJUyXk
Dec 15 13:54:19 aditi-VirtualBox usbguard-daemon[21576]: [1608026059.927] (A) uid=0 pid=21576 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:0b.0/usb1' target.new='allow' type='Policy.De
Dec 15 13:54:19 aditi-VirtualBox usbguard-daemon[21576]: [1608026059.927] (A) uid=0 pid=21576 device.rule='allow id 1d6b:0001 serial "0000:00:06.0" name "OHCI PCI host controller" hash "lUN32sIeMBBlD8Pd82
Dec 15 13:54:19 aditi-VirtualBox usbguard-daemon[21576]: [1608026059.927] (A) uid=0 pid=21576 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:06.0/usb2' target.new='allow' type='Policy.De
Dec 15 13:54:19 aditi-VirtualBox usbguard-daemon[21576]: [1608026059.928] (A) uid=0 pid=21576 device.rule='allow id 80ee:0021 serial "" name "USB Tablet" hash "8S88DbsXkyb93aEG099kxcbjrHSGfpZEJ8W0048wl1A=
Dec 15 13:54:19 aditi-VirtualBox usbguard-daemon[21576]: [1608026059.928] (A) uid=0 pid=21576 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:06.0/usb2/2-1' target.new='allow' type='Polic
Dec 15 14:16:35 aditi-VirtualBox systemd[1]: /lib/systemd/system/usbguard.service:29: Unknown system call group, ignoring: @system-service
```

If I stop the usbguard.service and start again this is the output:

```
aditi@aditi-VirtualBox:~$ sudo systemctl stop usbguard.service 
aditi@aditi-VirtualBox:~$ sudo systemctl status usbguard.service 
● usbguard.service - USBGuard daemon
   Loaded: loaded (/lib/systemd/system/usbguard.service; enabled; vendor preset: enabled)
[... 550 lines cut ...]

@timer
    # Schedule operations by time
    alarm
    getitimer
    setitimer
    timer_create
    timer_delete
    timer_getoverrun
    timer_gettime
    timer_settime
    timerfd_create
    timerfd_gettime
    timerfd_settime
    times
lines 412-458/458 (END)
```

There is no @system-service present in systemd 237.

If I comment out the line ```SystemCallFilter=@system-service``` in the usbguard.service.in file and follow the same installation and configuration steps, the USBGuard works perfectly.

I want to run USBGuard 0.7.8 on my Ubuntu 18.04 using systemd without removing the ```SystemCallFilter=@system-service```.

Please help me with the issue.

radosroka, 2021-01-15:
Hello, 

it looks like ubuntu's systemd problem. There's systemd-245.8-2.fc32.x86_64 on my Fedora system and it works.
If the current code works with the latest systemd it is sufficient. If I'm correct and this is ubuntu only problem because of old systemd then it should be fixed in ubuntu downstream.

aditiambadkar, 2021-01-15:
Yes, you are correct. After a lot of research even I came to the conclusion that it's a systemd version issue. USBGuard 0.7.8 works perfectly on Ubuntu 20.04 (systemd version 245 and in Ubuntu 18.04 systemd version is 237). Thank you for confirmation from your side.

closed 2021-01-15 by aditiambadkar