USBGuard/usbguard #444: Unable to temporarily allow device

liskin, 2020-12-27:
I'm on Debian testing, latest usbguard 0.7.8+ds-2. Whenever I `usbguard allow-device N`, the device gets authorized and deauthorized immediately after.

Here's the relevant part of `/usr/sbin/usbguard-daemon -d -k -c /etc/usbguard/usbguard-daemon.conf` output:
[usbguard1.log](https://github.com/USBGuard/usbguard/files/5744834/usbguard1.log)

The interesting bits:
```
[1609088503.468] (T) SysFSDevice.cpp@223/setAttribute: name=authorized value=1
[1609088503.468] (T) SysFSDevice.cpp@224/setAttribute: path=/devices/pci0000:00/0000:00:14.0/usb1/1-4
[1609088503.471] (D) UEventDeviceManager.cpp@484/thread: UEvent read event.
[1609088503.471] (T) UEvent.cpp@59/setAttribute: Setting attribute: ACTION=change
[1609088503.471] (T) UEvent.cpp@59/setAttribute: Setting attribute: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-4
…
[1609088503.487] (T) RulePrivate.cpp@113/appliesToWithConditions: return: applies=0
[1609088503.487] (T) Daemon.cpp@957/getPresentDevicePolicyRule: return: matched_rule=block
[1609088503.487] (T) Daemon.cpp@839/dmApplyDevicePolicy: device_ptr=0x55eb73f2e870 matched_rule_ptr=0x7f2754000cd0
[1609088503.487] (T) DevicePrivate.cpp@79/getDeviceRule: entry:  with_port=1 with_parent_hash=1 match_rule=0
[1609088503.487] (T) DeviceManagerPrivate.cpp@113/getDevice: id=27
[1609088503.487] (T) DevicePrivate.cpp@121/getDeviceRule: return: device_rule=allow id 0458:0186 serial "" name "Wired Mouse" hash "PQcWAl1T/zLuoroSVzVKStzzlkMrETuS88JQkZ7oLnA=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "1-4" with-interface { 03:01:02 03:00:00 } with-connect-type "hotplug"
[1609088503.487] (T) UEventDeviceManager.cpp@387/applyDevicePolicy: id=30 target=block
[1609088503.487] (T) DeviceManagerPrivate.cpp@113/getDevice: id=30
[1609088503.487] (T) SysFSDevice.cpp@223/setAttribute: name=authorized value=0
[1609088503.487] (T) SysFSDevice.cpp@224/setAttribute: path=/devices/pci0000:00/0000:00:14.0/usb1/1-4
[1609088503.575] (D) Daemon.cpp@850/dmApplyDevicePolicy: Device target changed: old=allow new=block
…
[1609088503.578] (W) USB Device Exception: SysFSDevice: (_sysfs_dirfd = open((G_sysfs_root + _sysfs_path).c_str(), O_PATH|O_DIRECTORY)) < 0: No such file or directory
[1609088503.578] (T) DeviceManagerPrivate.cpp@133/DeviceException: message=SysFSDevice: (_sysfs_dirfd = open((G_sysfs_root + _sysfs_path).c_str(), O_PATH|O_DIRECTORY)) < 0: No such file or directory
[1609088503.578] (W) SysFSDevice: (_sysfs_dirfd = open((G_sysfs_root + _sysfs_path).c_str(), O_PATH|O_DIRECTORY)) < 0: No such file or directory
```

So it seems that immediately after the device is authorized, kernel sends a change event and another usbguard thread processes that event and deauthorizes the device again. I'm sure this used to work a couple months ago, but it no longer does. I did upgrade the systemd numerous times since I last saw it working, so this might be a kernel change, might be a change in udev (https://github.com/systemd/systemd/blob/28423d9a75cf3e740997545ca13429a93b27181c/NEWS#L5 seems somewhat relevant), or something else entirely.

muelli, 2020-12-29:
>  did upgrade the systemd numerous times since I last saw it working, so this might be a kernel change, might be a change in udev (https://github.com/systemd/systemd/blob/28423d9a75cf3e740997545ca13429a93b27181c/NEWS#L5 seems somewhat relevant),

yeah, I was about to say that it reminds me of https://lwn.net/Articles/837033/, but it seems the BIND action has a dedicated handler: https://github.com/USBGuard/usbguard/blob/13c7256419027f0fd74d60fc9309ac4ead2f26f7/src/Library/UEventDeviceManager.cpp#L382
And your log shows that the action is not BIND. Or do you have any of those in your logs?

So, spontaneously, I don't think this is the cause of the issue. But can you downgrade your systemd (or Linux) to check whether it works as expected, there?

liskin, 2020-12-30:
There are some `bind` events later, but those should indeed be irrelevant as authorized is set to 0 during handling of the `change` event. The hypothesis was that the systemd change would somehow affect the timing or something.

I wasn't planning to reboot any time soon but I can try to reproduce this in a VM.

Minion3665, 2021-01-01:
I can reproduce this on manjaro 5.9.11-3, a workaround is to allow the device permanently with the `-p` flag but obviously, you then have to remember to remove the rule later.

liskin, 2021-01-01:
All right, I reproduced in a VM and the problem seems to be introduced in the 5.9 kernel:

* [usbguard-linux_4.19.160-systemd_241-usbguard_0.7.4.log](https://github.com/USBGuard/usbguard/files/5758894/usbguard-linux_4.19.160-systemd_241-usbguard_0.7.4.log): works ok, no `change` uevent emitted
* [usbguard-linux_4.19.160-systemd_247-usbguard_0.7.8.log](https://github.com/USBGuard/usbguard/files/5758895/usbguard-linux_4.19.160-systemd_247-usbguard_0.7.8.log): works ok, no `change` uevent emitted
* [usbguard-linux_5.8.10-systemd_247-usbguard_0.7.8.log](https://github.com/USBGuard/usbguard/files/5758896/usbguard-linux_5.8.10-systemd_247-usbguard_0.7.8.log): works ok, no `change` uevent emitted
* [usbguard-linux_5.9.15-systemd_247-usbguard_0.7.8.log](https://github.com/USBGuard/usbguard/files/5758897/usbguard-linux_5.9.15-systemd_247-usbguard_0.7.8.log): `change` uevent emitted immediately after writing to `authorized`, resulting in deauth

So I googled for what might be causing this and found 
https://bugzilla.kernel.org/show_bug.cgi?id=209469. I'm moderately certain https://github.com/torvalds/linux/commit/a45aca510b73b745f27f39c6bb590b1743ea1792 is the culprit here as well.

We could report this to the kernel devs and invoke the "never break userspace" rule, but I took a glance over usbguard sources and it seems to me the device would be deathorized even if the `change` uevent wasn't emitted immediately but instead arrived a couple seconds/minutes/hours later for some other reason, which it might, because `usbguard allow-device` invokes `IPCServerPrivate::handleApplyDevicePolicy` but doesn't append the rule anywhere for other policy applications to see (I may be totally wrong about this, of course). So the better fix would be for `usbguard allow-device` to first append the rule (with `permanent=false`) and only then apply the policy.

Cropi, 2021-01-04:
There is an ongoing PR, which fixes this issue (https://github.com/USBGuard/usbguard/pull/440). However, there might be some unexpected consequences.

radosroka, 2021-01-15:
> There is an ongoing PR, which fixes this issue (#440). However, there might be some unexpected consequences.

PR was merged and there is a new upstream version(1.0.0) available which contains the commit https://github.com/USBGuard/usbguard/commit/9b881ed208c9f8ca8515d69aa328dad79708b778.

closed 2021-01-15 by radosroka

liskin, 2021-01-15:
Yeah, I meant to close this and forgot about it. I tested the fix in a VM and it works, so thank you!