USBGuard/usbguard #447: Feature request: Allow disabling usbguard with a boot parameter

zivanfi, 2021-01-19:
I recently stumbled upon usbguard. I really liked the idea at first but then it has occurred to me that without a USB keyboard I could easily lose access to my computer. I found #246 which describes such a case. This could even happen a long time after installation if my keyboard breaks and I have no backup keyboard whitelisted and at hand.

Booting from a live distro, chowning into the main installation and fixing it manually is (hopefully) always possible but is a hassle with complex setups such as ones involving lvm, luks or exotic filesystems. I would suggest a safety switch instead in case one wants to override usbguard.

There is an opportunity for user input before usbguard even has a chance to start: the boot manager menu. This usually not only allows selecting one of the preconfigured options but also editing the kernel parameters as well. After bootup, these kernel parameters become easily accessible to userspace applications too in the form of the /proc/cmdline file.

Usbguard could check the kernel parameters for a magic string upon startup and exit without blocking any USB devices if the magic string is present there. This would allow users to disable usbguard from the boot menu. A hardcoded magic string would suffice in my opinion, but if a malicious USB device navigating its way around in GRUB to add this kernel parameter is a concern then it could even be made configurable by the user for improved security.

If you find this idea feasible and useful, please consider implementing it. Thanks!

AnanasPfirsichSaft, 2021-01-31:
For the record: You can lock your GRUB configuration with a password. It has to be entered before any changes can be made in the GRUB menu. So even a hardcoded magic key cannot be set by bad hardware.

If you use systemd, maybe a simple "systemd.mask=usbguard.service" would suffice?

zivanfi, 2021-01-31:
`systemd.mask=usbguard.service` sounds like the perfect solution, thanks! I tried it and it works like a charm.

(May I suggest documenting this as a rescue procedure though? Based on the reports I've seen on the internet, there are many users who managed to lock themselves out of their computers with usbguard and had to resort to fixing their systems using a live distro, which is a lot more work than specifying `systemd.mask=usbguard.service` at the boot menu.)

closed 2021-01-31 by zivanfi

muelli, 2021-02-01:
Does it also work when USBGuard is accessed through DBus? I would expect the daemon to be started once somebody tries to access it.