USBGuard/usbguard #450: [1.0.0] 5 tests fail for ./configure --enable-full-test-suite

hartwork, 2021-02-09:
These 5 tests fail for me:
```
FAIL: UseCase/001_cli_policy.sh
FAIL: Source/check-driver.sh
FAIL: UseCase/003_cli_devices_umockdev.sh
FAIL: UseCase/004_daemonize.sh
FAIL: UseCase/005_cli_devices_advanced.sh
```
Here are the details from the log:
```
==============================================
   usbguard 1.0.0: src/Tests/test-suite.log
==============================================

# TOTAL: 11
# PASS:  5
# SKIP:  1
# XFAIL: 0
# FAIL:  5
# XPASS: 0
# ERROR: 0

.. contents:: :depth: 2

FAIL: Source/check-driver.sh
============================

All source files should include build-config.h before any other file, add: 
--- snip ---
 #ifdef HAVE_BUILD_CONFIG_H
   #include <build-config.h>
 #endif
--- snip ---
to the beginning of the source file.

File: src/build-config.h
                copyright.sh: FAILED!
private-with-build-config.sh: FAILED!
             vim-modeline.sh: FAILED!

All source files should include build-config.h before any other file, add: 
--- snip ---
 #ifdef HAVE_BUILD_CONFIG_H
   #include <build-config.h>
 #endif
--- snip ---
to the beginning of the source file.

File: src/Library/IPC/Policy.pb.h
                copyright.sh: FAILED!
private-with-build-config.sh: FAILED!
             vim-modeline.sh: FAILED!

All source files should include build-config.h before any other file, add: 
--- snip ---
 #ifdef HAVE_BUILD_CONFIG_H
   #include <build-config.h>
 #endif
--- snip ---
to the beginning of the source file.

File: src/Library/IPC/Parameter.pb.h
                copyright.sh: FAILED!
private-with-build-config.sh: FAILED!
             vim-modeline.sh: FAILED!

All source files should include build-config.h before any other file, add: 
--- snip ---
 #ifdef HAVE_BUILD_CONFIG_H
   #include <build-config.h>
 #endif
--- snip ---
to the beginning of the source file.

File: src/Library/IPC/Exception.pb.h
                copyright.sh: FAILED!
private-with-build-config.sh: FAILED!
             vim-modeline.sh: FAILED!

All source files should include build-config.h before any other file, add: 
--- snip ---
 #ifdef HAVE_BUILD_CONFIG_H
   #include <build-config.h>
 #endif
--- snip ---
to the beginning of the source file.

File: src/Library/IPC/Devices.pb.h
                copyright.sh: FAILED!
private-with-build-config.sh: FAILED!
             vim-modeline.sh: FAILED!

All source files should include build-config.h before any other file, add: 
--- snip ---
 #ifdef HAVE_BUILD_CONFIG_H
   #include <build-config.h>
 #endif
--- snip ---
to the beginning of the source file.

File: src/Library/IPC/Rule.pb.h
                copyright.sh: FAILED!
private-with-build-config.sh: FAILED!
             vim-modeline.sh: FAILED!

All source files should include build-config.h before any other file, add: 
--- snip ---
 #ifdef HAVE_BUILD_CONFIG_H
   #include <build-config.h>
 #endif
--- snip ---
to the beginning of the source file.

File: src/Library/IPC/Message.pb.h
                copyright.sh: FAILED!
private-with-build-config.sh: FAILED!
             vim-modeline.sh: FAILED!

FAIL Source/check-driver.sh (exit status: 1)

FAIL: UseCase/001_cli_policy.sh
===============================

/usr/bin/umockdev-wrapper
[1612882112.624] (W) PERMISSIONS CHECK ON POLICY FILE ARE TURNED OFF!
[1612882112.625] (E) IPC server initialization: service: Address already in use
ERROR: IPC connect: service=usbguard: Operation not permitted
ERROR: IPC connect: service=usbguard: Operation not permitted
../../src/Tests/bash-testlib.sh: line 189: kill: (26567) - No such process
FAIL UseCase/001_cli_policy.sh (exit status: 1)

SKIP: UseCase/002_cli_devices.sh
================================

ERROR: ld.so: object 'libsandbox.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
sudo: a password is required
ERROR: ld.so: object 'libsandbox.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
sudo: a password is required
SKIP UseCase/002_cli_devices.sh (exit status: 77)

FAIL: UseCase/003_cli_devices_umockdev.sh
=========================================

/usr/bin/umockdev-wrapper
[1612882114.680] (W) PERMISSIONS CHECK ON POLICY FILE ARE TURNED OFF!
[1612882114.680] (E) IPC server initialization: service: Address already in use
[1612882118.560] (T) IPCClientPrivate.cpp@117/connect: 
[1612882118.561] (T) IPCClientPrivate.cpp@103/destruct: 
ERROR: IPC connect: service=usbguard: Operation not permitted
../../src/Tests/bash-testlib.sh: line 189: kill: (27063) - No such process
FAIL UseCase/003_cli_devices_umockdev.sh (exit status: 1)

FAIL: UseCase/004_daemonize.sh
==============================

[1612882115.338] (W) PERMISSIONS CHECK ON POLICY FILE ARE TURNED OFF!
[1612882115.339] (E) IPC server initialization: service: Address already in use
../../src/Tests/bash-testlib.sh: line 189: kill: (27260) - No such process
cat: /var/tmp/portage/sys-apps/usbguard-1.0.0/temp/usbguard-test.byIRSz/usbguard.pid: No such file or directory

Usage:
 kill [options] <pid> [...]

Options:
 <pid> [...]            send signal to every <pid> listed
 -<signal>, -s, --signal <signal>
                        specify the <signal> to be sent
 -l, --list=[<signal>]  list all signal names, or convert one to a name
 -L, --table            list all signal names in a nice table

 -h, --help     display this help and exit
 -V, --version  output version information and exit

For more details see kill(1).
FAIL UseCase/004_daemonize.sh (exit status: 1)

FAIL: UseCase/005_cli_devices_advanced.sh
=========================================

/usr/bin/umockdev-wrapper
[1612882117.265] (W) PERMISSIONS CHECK ON POLICY FILE ARE TURNED OFF!
[1612882117.266] (E) IPC server initialization: service: Address already in use
[1612882121.251] (T) IPCClientPrivate.cpp@117/connect: 
[1612882121.252] (T) IPCClientPrivate.cpp@103/destruct: 
ERROR: IPC connect: service=usbguard: Operation not permitted
../../src/Tests/bash-testlib.sh: line 189: kill: (27985) - No such process
FAIL UseCase/005_cli_devices_advanced.sh (exit status: 1)
```

Cropi, 2021-02-11:
Hi @hartwork,

I executed the tests locally but I did not encounter any issues. Just a quick note - in case you want to execute all the tests, make sure that no other usbguard-daemon instance is running. According to the logs you provided, there is another instance running, so that's why the tests failed.

hartwork, 2021-02-11:
> in case you want to execute all the tests, make sure that no other usbguard-daemon instance is running. According to the logs you provided, there is another instance running, so that's why the tests failed.

@Cropi good to know, thanks! I did that now — stopped the local running deamon (and also ran things outside of the sandbox)  — and that made all tests work _but_ test `Source/check-driver.sh`. I have astyle 3.1 here, if that matters. The fail log is this:

```console
# cat src/Tests/test-suite.log
==============================================
   usbguard 1.0.0: src/Tests/test-suite.log
==============================================

# TOTAL: 11
# PASS:  9
# SKIP:  1
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0

.. contents:: :depth: 2

FAIL: Source/check-driver.sh
============================

All source files should include build-config.h before any other file, add: 
--- snip ---
 #ifdef HAVE_BUILD_CONFIG_H
   #include <build-config.h>
 #endif
--- snip ---
to the beginning of the source file.

File: src/build-config.h
             vim-modeline.sh: FAILED!
private-with-build-config.sh: FAILED!
                copyright.sh: FAILED!

All source files should include build-config.h before any other file, add: 
--- snip ---
 #ifdef HAVE_BUILD_CONFIG_H
   #include <build-config.h>
 #endif
--- snip ---
to the beginning of the source file.

File: src/Library/IPC/Parameter.pb.h
             vim-modeline.sh: FAILED!
private-with-build-config.sh: FAILED!
                copyright.sh: FAILED!

All source files should include build-config.h before any other file, add: 
--- snip ---
 #ifdef HAVE_BUILD_CONFIG_H
   #include <build-config.h>
 #endif
--- snip ---
to the beginning of the source file.

File: src/Library/IPC/Policy.pb.h
             vim-modeline.sh: FAILED!
private-with-build-config.sh: FAILED!
                copyright.sh: FAILED!

All source files should include build-config.h before any other file, add: 
--- snip ---
 #ifdef HAVE_BUILD_CONFIG_H
   #include <build-config.h>
 #endif
--- snip ---
to the beginning of the source file.

File: src/Library/IPC/Exception.pb.h
             vim-modeline.sh: FAILED!
private-with-build-config.sh: FAILED!
                copyright.sh: FAILED!

All source files should include build-config.h before any other file, add: 
--- snip ---
 #ifdef HAVE_BUILD_CONFIG_H
   #include <build-config.h>
 #endif
--- snip ---
to the beginning of the source file.

File: src/Library/IPC/Devices.pb.h
             vim-modeline.sh: FAILED!
private-with-build-config.sh: FAILED!
                copyright.sh: FAILED!

All source files should include build-config.h before any other file, add: 
--- snip ---
 #ifdef HAVE_BUILD_CONFIG_H
   #include <build-config.h>
 #endif
--- snip ---
to the beginning of the source file.

File: src/Library/IPC/Rule.pb.h
             vim-modeline.sh: FAILED!
private-with-build-config.sh: FAILED!
                copyright.sh: FAILED!

All source files should include build-config.h before any other file, add: 
--- snip ---
 #ifdef HAVE_BUILD_CONFIG_H
   #include <build-config.h>
 #endif
--- snip ---
to the beginning of the source file.

File: src/Library/IPC/Message.pb.h
             vim-modeline.sh: FAILED!
private-with-build-config.sh: FAILED!
                copyright.sh: FAILED!

FAIL Source/check-driver.sh (exit status: 1)

SKIP: UseCase/002_cli_devices.sh
================================

sudo: a password is required
sudo: a password is required
SKIP UseCase/002_cli_devices.sh (exit status: 77)
```

Any ideas why this is happening?

Cropi, 2021-02-11:
Some of the tests require root privileges. If I am not mistaken, this particular test adds/removes modules from the linux kernel(```modprobe, rmmod```). Try to use sudo.

hartwork, 2021-02-11:
I think you're referring to `UseCase/002_cli_devices.sh` rather than `Source/check-driver.sh`. if I apply this diff…
```diff
# git --no-pager diff -U0
diff --git a/src/Tests/Makefile.am b/src/Tests/Makefile.am
index 4ef2310..e773850 100644
--- a/src/Tests/Makefile.am
+++ b/src/Tests/Makefile.am
@@ -86 +85,0 @@ TESTS+=\
-       UseCase/002_cli_devices.sh \
```
…and re-run tests, I get no more warnings about sudo but `Source/check-driver.sh` still fails (with a subset of the log I pasted above).

ZoltanFridrich, 2021-02-12:
@Cropi modprobe is used within 002_cli_devices.sh and that test is practically unusable because of the missing module for simulating USB devices on fedora which is no longer present by default. But that's not the issue here because 002_cli_devices.sh is intentionally skipped.

The test which is failing is Source/check-driver.sh which checks whether each source file includes build-config.h. However as is apparent from the logs, it also checks files .pb.h which is a source code generated by protocol buffer and therefore obviously does not include build-config.h, also the test checks build-config.h itself which is an obvious error.

@hartwork can you confirm that this PR fixes the issue?
https://github.com/USBGuard/usbguard/pull/452

hartwork, 2021-02-12:
> @hartwork can you confirm that this PR fixes the issue? #452

I confirm that this fixes #450 for me, very nice! tada pray

PR #452 (2021-02-18): Fix test Source/check-driver.sh
Resolves #450
closed 2021-02-18 by radosroka