USBGuard/usbguard #456: Rule allowed-match and cannot resolve realpath

reznik72, 2021-02-25:
Hello,

I has been trying write usbguard rule for allowing one USB Flash disk by defining only ID and NAME, only if the same device is not allowed.

This is the rule: **allow id 090c:3267 name "Flash Disk" if !allowed-matches(id 090c:3267 name "Flash Disk")**

If the USB device is not inserted when the usbguard daemon is started, everything is OK and works as expected. But if the USB device is inserted before the usbguard daemon is started, it has unexpected behaviour for me, because at first start, everything is OK, then I restart the usbguard daemon and the USB device is blocked (yes, it is inserted only one USB device in PC) and usbguard daemon outputs this warning message: **(W) Cannot resolve realpath for /sys/bus/usb/devices/../../../devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9:1.0**. Then I restart usbguard daemon again and USB device is allowed and everything works, then I restart usbaguard again and there is same problem and thus it is repeated.

So it seems if the in one running instance of usbguard the USB device was allowed, then in next running instance is shown this error.

I tested it on Debian 10 with usbguard 0.7.8 and also 1.0.0.

ZoltanFridrich, 2021-03-04:
There is no bug here, it is working as expected. However, I agree that it is not very intuitive.

What actually happens is this: (assuming that PresentDevicePolicy=apply-policy)
**1.** device is **blocked** before usbguard is started:
After the daemon is started it will scan for present USB devices. After scanning through your "Flash Disk" device, it will be added to the list of devices (target=block). Then the daemon will try to apply the policy to this device and obviously, it will get matched with your rule because this device is not allowed. Therefore this device gets allowed as expected.
**2.** device is **allowed** before usbguard is started:
Again, after the daemon is started, devices are scanned. "Flash Disk" device is scanned and will be added to the list of devices (target=allow). Then the daemon will try to apply the policy to this device but because DeviceManager already knows this device and it is allowed, then this device will not match your rule and if there is no other rule that matches the device then ImplicitPolicyTarget will be applied (block by default). Then scanning continues with children's paths of the "Flash Disk" device but because this device has been blocked, the paths will be invalid, and "Cannot resolve realpath warning" will be generated. (This warning is not harmful and can be ignored)

reznik72, 2021-03-04:
Ok, thank you for explanation. Could you advise me, how to write this rule or if it is even possible?

I need rule for allowing one USB Flash disk, by defining only ID and NAME, only if the same device is not allowed.

Now I solved this problem in daemon which listens on IPC, but it would be more elegant, if there is some rule for this.

ZoltanFridrich, 2021-03-04:
I might be wrong, but this should be what you want,try this rule:
`allow id 090c:3267 name "Flash Disk" if one-of { !rule-applied !allowed-matches(id 090c:3267 name "Flash Disk") }`