USBGuard/usbguard #457: Connection failed: IPC connect - how to allow all currently connected USB devices and start usbguard.service?

mYnDstrEAm, 2021-03-02:
Like in #161 I'm getting this error in Debian10/KDE:

```
service=usbguard: Connection refused
Connection failed: IPC connect:
```

When running `sudo systemctl start usbguard.service` as recommended in the linked issue all connected USB devices, including the keyboard, stop working.

How to allow all currently connected USB devices?

Furthermore, this problem should not appear after installing & running UsbGuard on Debian10 and the usbguard.service should start automatically except if another setting is chosen in the app's config. Once all currently connected USB devices are allowed, I'm interested in where to find a list of the allowed devices and get KDE notifications for an allow/reject prompt whenever a new USB device gets connected.

Cropi, 2021-03-03:
Hi @mYnDstrEAm ,

Before you start the usbguard system service, make sure that you have your policy generated. You can do that by using CLI:
```
usbguard generate-policy
```
Place the output to your rules.conf(edit it according to your needs) file then you can safely start the service with ```sudo systemctl start usbguard.service```

hartwork, 2022-01-28:
I think this ticket may be a duplicate of #161. Should it be closed?

radosroka, 2022-01-28:
Yeah, this is quite old. Let's close it.

closed 2022-01-28 by radosroka