USBGuard/usbguard #460: usbguard hardening

glitsj16, 2021-03-08:
To be clear, this is _not_ a bug report, more of _logic check_ or a _feature-request_.

I've been using usbguard (manually built from git) for quite some time, both on Arch Linux and Ubuntu LTS machines. In an attempt to harden my systemd services as tight as possible I noticed usbguard-daemon runs as root and the usbguard.service uses a PIDFile. Both are not optimal IMO (I am aware of the -C switch though).

Recently systemd began [stressing](https://github.com/systemd/systemd/commit/5cfbe7452f023fc307374f306e26b5ee902889f0) that PID files should be avoided in modern projects, advising to use Type=notify or Type=simple to avoid needless forking. Running usbgard `without` the PIDFile=/run/usbguard.pid and as `Type=simple` _seems_ to work just fine when using `Execstart=/usr/bin/usbguard-daemon -s -c /etc/usbguard/usbguard-daemon.conf` (so dropping -f). What is the rationale behind using Type=forking currently?

Trying to run usbguard-daemon as a unprivileged, non-root user seems _impossible_ with the current code base. systemd's `DynamicUser` concept or the more traditional /usr/lib/sysusers.d/foo.conf mechanism both fail due to what I think are the in-built permissions/ownership checks on dirs/files under /etc/usbguard. I sure don't want to use the -P switch to drop these checks, but I've tried and even then the service goes into failure mode. Is there a known, documented way of running usbguard-daemon as unprivileged user (e.g. usbguard)? If not, is such a feature planned or does it make no sense to even consider such a scenario?

radosroka, 2021-03-08:
Here is the commit a2f89d6edcb4a3d09d5108ef6ca03d88e2971713 that changed type from`simple` to `forking`.

- with such configuration "systemctl start usbguard" can return error code in case of bad configuration

 That is the reason of the change.

glitsj16, 2021-03-08:
@radosroka Thanks for pointing that out, appreciated. I'll do some more experimenting but I think it's best to close this for now.

closed 2021-03-08 by glitsj16