USBGuard/usbguard #461: Feature Request: Enable comments in rules.conf

ckastner, 2021-03-14:
A way to add comments to the rules file syntax would be nice, eg: "any line beginning with a hash sign is treated as a comment".

Cropi, 2021-03-15:
Hi @ckastner,

It is already part of usbguard since 0.7.5 release.

ckastner, 2021-03-15:
Apologies, it appears that I tested this with Debian stable's 0.7.4 instead of 1.0.0 I had installed on another system.

May I suggest a short one-line modification to `usbguard-rules.conf.5.adoc`? I'd propose a PR, but I'm not sure where you'd prefer the comment (before, in, after the rule spec).

Cropi, 2021-03-16:
It looks like that comments are not described in the manual pages. It would be a great idea to somehow include it in the documentation. I think it would make sense if it was part of the _rule language_ section, inside `usbguard-rules.conf.5.adoc` file.

Yes, we do appreciate PRs :) Thanks.

hartwork, 2022-02-19:
Please note issue "Permanently allowing device deletes comments in rules.conf" (#486)

calestyo, 2022-11-01:
So given that this is not yet documented… how are comments made?

hartwork, 2022-11-01:
@calestyo the character starting comments in rules syntax is `#`, a hash mark.

(For proof:) https://github.com/USBGuard/usbguard/blob/927a41da7538d9f31846bdf14a49197d8e5a0d80/src/Library/RuleParser/Grammar.hpp#L207

hartwork, 2022-11-01:
PS: Also there is:
https://github.com/USBGuard/usbguard/blob/927a41da7538d9f31846bdf14a49197d8e5a0d80/src/Tests/Rules/test-rules.good#L767-L770

hartwork, 2022-12-01:
@Cropi I would like to suggest this ticket either for closing as "fixed, already works" or to pin it, so that it's most easy to find. What do you think?

calestyo, 2022-12-01:
Can't we just add a line the `usbguard-rules.conf` telling how comments look like?

hartwork, 2022-12-01:
@calestyo good point, I guess I forgot about https://github.com/USBGuard/usbguard/issues/461#issuecomment-800143546 above :+1:

calestyo, 2022-12-07:
Thanks, but is that written clearly enough?

Is really **everything** after a `#` ignored? Or just when there was only leading (or no) whitespace? And if it's actually the former, wouldn't one need a way to quote a `#` if it shall e.g. be part of a device name or so?

Cropi, 2022-12-08:
> Thanks, but is that written clearly enough?
> 
> Is really **everything** after a `#` ignored? Or just when there was only leading (or no) whitespace? And if it's actually the former, wouldn't one need a way to quote a `#` if it shall e.g. be part of a device name or so?

One or more leading white spaces should already be handled properly. I think that the symbol is not treated as a comment if it's inside the rule, although will need to double check it. If I am wrong on this, can you create a new issue for this?

commit 32713be4 (2022-12-07): Describe comments in the manual page, fixes #461 (#572)