USBGuard/usbguard #472: "Device is not authorized for usage" when connecting new USB devices after stopping usbguard-daemon

wjf3121, 2021-04-14:
Hi all,

What's the expected behavior when stopping usbguard daemon? I started usbguard daemon using command "sudo usbguard-daemon -d". It worked as expected -- all rules were applied. However, when I killed the daemon server and connected a new USB device to my desktop, dmesg showed "Device is not authorized for usage". I could only connect the new device after rebooting the desktop.

I expect any USB devices will be allowed when usbguard daemon is killed. Is this not the case? Thanks!

Cropi, 2021-04-14:
Hi,

It's just an assumption but it might be connected with the ```AuthorizedDefault``` configuration option. By default, it is set to ```none```. It enables you to define what value the default authorization is set to. In case you insert a usb new device and usbguard is not running but the value of authorized_default is set to 0 -> the device won't be authorized.

wjf3121, 2021-04-14:
Thanks Cropi! Setting AuthorizedDefault to all does solve this issue. Really appreciate it:-)

closed 2021-04-14 by wjf3121