USBGuard/usbguard #473: Block vs Reject

molitona, 2021-04-15:
Hi, I wonder what's the deep difference between Block and Reject (I've already seen the do
[image]
?

Also, do they both prevent USBKiller ?
And do they both prevent the other malicious USB's (those who exfiltrate data) ?

Cropi, 2021-04-19:
Hi @molitona,

The key difference is that:
-  ```block``` will deauthorize the device and it will stay in a blocked state until it is authorized.
- ```reject``` will deauthorize **and remove** the device. In case you would like to use a rejected device again, you will have to reinsert it.

Both should protect against malicious USBs by a set of usbguard rules.

muelli, 2021-04-19:
But USBKiller is the name of a device that has too high voltage for USB, no? In that case, any software based solution is not capable of defending against such an attack.

molitona, 2021-04-19:
Oh, unfortunately. Thanks!

-------- Original Message --------

molitona, 2021-04-19:
Then, usbguard can only protect software attacks (those who exfiltrate data), right ?

-------- Original Message --------

Cropi, 2021-04-21:
USBGuard was designed to authorize/deauthorize/reject usb devices by a set of rules. It's up to the user to decide, which devices are authorized for use. To sum up, deauthorized devices can not exfiltrate data from your machine.

molitona, 2021-04-21:
Ok, thanks!

-------- Original Message --------

closed 2021-04-21 by Cropi