USBGuard/usbguard #476: usbguard-applet-qt missing in Debian Testing (bullseye)

AreYouLoco, 2021-05-06:
What happened to the nice gui of usbguard? I've read somewhere that it was merged into usbguard but cannot install/compile it for Debian Bullseye. What happened?

AreYouLoco, 2021-05-06:
I see that in latest source there is missing --with-gui-qt=qt5 option that prevents from building applet.

genodeftest, 2021-06-03:
It has been removed from usbguard 0.7.5, see the [release notes](https://github.com/USBGuard/usbguard/releases/tag/usbguard-0.7.5) and it will not come back.

According to the [official announcement](https://usbguard.github.io/blog/2019/usbguard-0.7.6) you should use [usbguard-notifier](https://github.com/Cropi/usbguard-notifier) now. There is also some very basic integration in GNOME 40.

MatejKovacic, 2021-07-30:
Are there any plans for integration of `usbguard-notifier` applet into the `usbguard`? It would be nice if people would be able to use GUI... I am sure there will be much more users for that great software.

hartwork, 2022-02-19:
Based on https://github.com/USBGuard/usbguard/issues/476#issuecomment-853766678 I believe this ticket can be closed as "wont fix".

closed 2022-02-23 by radosroka