USBGuard/usbguard #485: error while loading shared libraries: libusbguard.so.1

TriplEight, 2021-07-30:
Even after `sudo apt install --reinstall usbguard` I'm getting
```
$ sudo usbguard list-devices                                                                                                                                                                                        
usbguard: error while loading shared libraries: libusbguard.so.1: cannot open shared object file: No such file or directory
$ sudo apt install --reinstall usbguard                                                                                                                                                                                    
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 140 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://de.archive.ubuntu.com/ubuntu hirsute/universe amd64 usbguard amd64 1.0.0+ds-2 [140 kB]
Fetched 140 kB in 0s (317 kB/s)
(Reading database ... 412074 files and directories currently installed.)
Preparing to unpack .../usbguard_1.0.0+ds-2_amd64.deb ...
Unpacking usbguard (1.0.0+ds-2) over (1.0.0+ds-2) ...
Setting up usbguard (1.0.0+ds-2) ...
usbguard-dbus.service is a disabled or a static unit not running, not starting it.
Processing triggers for dbus (1.12.20-1ubuntu3) ...
Processing triggers for man-db (2.9.4-2) ...
 $ sudo usbguard list-devices                                                                                                                                                                                           
usbguard: error while loading shared libraries: libusbguard.so.1: cannot open shared object file: No such file or directory

brpatel12, 2021-12-20:
Having the same issue on Ubuntu 20.04

brpatel12, 2021-12-21:
@TriplEight 
```
./configure LDFLAGS="-static" --with-crypto-library=sodium --with-bundled-catch --with-bundled-pegtl --enable-static
make
sudo make install
```

pcraciunoiu, 2021-12-28:
Seeing this as well, works from build directory. It seems like `libusbguard.so.1` is not where it's expected to be, 

Without installing the apt package, I got this error trying to launch the build:

```
$ sudo ./usbguard list-devices
ERROR: IPC connect: service=usbguard: Connection refused
```

But if I `sudo apt install usbguard` to get the latest version the `./usbguard` executable works. Pretty strange. Would be great to get some help here!

Cropi, 2022-01-03:
Connection is mostly refused when the user/group you are using is not listed in either
- ```IPCAllowedGroups``` 
- ```IPCAllowedUsers```
- ```IPCAccessControlFiles```

For additional information, see the ```usbguard-daemon.conf``` manual page, section IPC access control. Let me know if that helps.

> Seeing this as well, works from build directory. It seems like `libusbguard.so.1` is not where it's expected to be,
>
> Without installing the apt package, I got this error trying to launch the build: [...]

pcraciunoiu, 2022-01-03:
Hey @Cropi thanks for your help.

That doesn't help. I tried creating those groups and adding root to it. Also looking in `usbguard-daemon.conf` in the cloned repo/build shows that `root` is listed as an allowed user anyway. But I get the same error.

Cropi, 2022-01-04:
Hey @pcraciunoiu,

Please try to not mix the files built from sources and those installed through the distribution you are using. By default, ```sudo make install``` command installs all the necessary files to ```/usr/local```, so your usbguard configuration files will be located inside this directory. However, I think the linker path is not updated after that(judging from the error report). It can be addressed by ```ldconfig /usr/local/lib/```.

pcraciunoiu, 2022-01-04:
@Cropi that fixes this error:

```
usbguard: error while loading shared libraries: libusbguard.so.1: cannot open shared object file: No such file or directory
```

However, the connection refused error remains. Now it happens consistently between the local usbguard (from the build) and the global one added by `sudo make install`:

```
➜  usbguard git:(b9ea78e) ✗ sudo ./usbguard list-devices
ERROR: IPC connect: service=usbguard: Connection refused
➜  usbguard git:(b9ea78e) ✗ sudo usbguard list-devices  
ERROR: IPC connect: service=usbguard: Connection refused
```

Any ideas? The linux groups for those IPC permissions are not created automatically by the build script, but yesterday I tried creating them and adding the root user to them, and that didn't solve the issue.

The distro package seems to fix this, but to follow what you suggested, I uninstalled it and ran `sudo make install` to make sure I'm not mixing the two.

nickt28, 2024-07-07:
Thanks, running `sudo ldconfig /usr/local/lib/` worked for me. I was getting an error when generating the rules file.

```
user@user-pf:~$ sudo sh -c 'usbguard generate-policy > /etc/usbguard/rules.conf'
usbguard: error while loading shared libraries: libusbguard.so.1: cannot open shared object file: No such file or directory
```