USBGuard/usbguard #496: run "# ./configure ..." failed

lilyeyes, 2021-11-10:
Hi experts,
I am a QE I would like to test the usbguard but I hit some issue, please help me out, thanks:
1. downloaded the ZIP (CODE -> Download ZIP) file from https://github.com/USBGuard/usbguard

2. unzip the ZIP file

3. check the files:
```
/test/usbguard-master # ll
total 164
-rw-r--r-- 1 root root    35 Sep 27 06:13 .gitattributes
drwxr-xr-x 1 root root    18 Sep 27 06:13 .github
-rw-r--r-- 1 root root   295 Sep 27 06:13 .gitignore
-rw-r--r-- 1 root root   211 Sep 27 06:13 .gitmodules
-rw-r--r-- 1 root root   475 Sep 27 06:13 .lgtm.yml
-rw-r--r-- 1 root root 18244 Sep 27 06:13 CHANGELOG.md
-rw-r--r-- 1 root root 18027 Sep 27 06:13 LICENSE
-rw-r--r-- 1 root root 17669 Sep 27 06:13 Makefile.am
-rw-r--r-- 1 root root  3419 Sep 27 06:13 README.adoc
-rw-r--r-- 1 root root  1481 Sep 27 06:13 RELEASE_PROCESS.md
-rw-r--r-- 1 root root     6 Sep 27 06:13 VERSION
drwxr-xr-x 1 root root    36 Sep 27 06:13 abi-ref
-rwxr-xr-x 1 root root   841 Sep 27 06:13 autogen.sh
-rwxrwxrwx 1 root root 26206 Nov  9 23:46 configure.ac
-rwxr-xr-x 1 root root 26206 Nov  9 23:47 configure.ac.bk
drwxr-xr-x 1 root root     6 Sep 27 06:13 doc
-rw-r--r-- 1 root root   274 Sep 27 06:13 libusbguard.pc.in
drwxr-xr-x 1 root root    98 Sep 27 06:13 m4
drwxr-xr-x 1 root root   284 Sep 27 06:13 scripts
drwxr-xr-x 1 root root    98 Sep 27 06:13 src
-rw-r--r-- 1 root root  6676 Sep 27 06:13 usbguard-daemon.conf.in
-rw-r--r-- 1 root root   883 Sep 27 06:13 usbguard.service.in
```
4. install some of the pkgs mentioned in README.adoc "Compilation & Installation"

5. I don't found the ./configure file so I tried configure.ac instead and the following cmds failed (see bellow)

6. # ./configure.ac --with-crypto-library=sodium
./configure.ac: line 1: syntax error near unexpected token `[2.69]'
./configure.ac: line 1: `AC_PREREQ([2.69])'

7.
susetest:/test/usbguard-master # make
make: *** No targets specified and no makefile found.  Stop.

8.
susetest:/test/usbguard-master # make check
make: *** No rule to make target 'check'.  Stop.

ZoltanFridrich, 2021-11-10:
you need to run ./autogen.sh first to generate configure script

lilyeyes, 2021-11-10:
Thank you for your quick replying.
I am trying. I hit some issues some of them may be introduced by lacking of needed pkgs in our OS.

Cropi, 2021-11-12:
Dependencies are listed in the README, let us know if there is something missing.

closed 2021-11-12 by Cropi