USBGuard/usbguard #499: Output From `generate-policy` Command Results In Error `parse error matching tao::pegtl::sor<usbguard::RuleParser::comment, usbguard::RuleParser::rule

bonedaddy, 2021-12-19:
# Overview

```
⋊> ~ sudo usbguard generate-policy > /etc/usbguard/rules.conf                                                                                                                         13:32:31
⋊> ~ sudo systemctl restart usbguard                                                                                                                                                  13:32:36
Job for usbguard.service failed because the control process exited with error code.
See "systemctl status usbguard.service" and "journalctl -xeu usbguard.service" for details.
⋊> ~ sudo journalctl -xeu usbguard | tail -n 60 | head -n 10                                                                                                                          13:32:46
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit usbguard.service has begun execution.
░░ 
░░ The job identifier is 8945.
Dec 19 13:32:45 biggray usbguard-daemon[167656]: [1639949565.173] (E) Rules: SourceLOCAL::/etc/usbguard/rules.conf: usbguard::Exception
Dec 19 13:32:45 biggray usbguard-daemon[167656]: Rules: SourceLOCAL::/etc/usbguard/rules.conf: usbguard::Exception
Dec 19 13:32:45 biggray systemd[1]: usbguard.service: Control process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
⋊> ~ usbguard-rule-parser /etc/usbguard/rules.conf                                                                                                                                    13:32:47
INPUT: /etc/usbguard/rules.conf
! ERROR: RuleParserError
!!  /etc/usbguard/rules.conf
!!  ^-- <argv>:1:0(0): parse error matching tao::pegtl::sor<usbguard::RuleParser::comment, usbguard::RuleParser::rule>
```

# Version Information

* /etc/os-release

```
NAME="Pop!_OS"
VERSION="21.10"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 21.10"
VERSION_ID="21.10"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=impish
UBUNTU_CODENAME=impish
LOGO=distributor-logo-pop-os
```

* `usbguard` package information

```
⋊> ~ sudo apt info usbguard                                                                                                                                                           13:33:31
Package: usbguard
Version: 1.0.0+ds-2
Priority: optional
Section: universe/utils
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Birger Schacht <birger@debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 518 kB
Pre-Depends: init-system-helpers (>= 1.54~)
Depends: libaudit1 (>= 1:2.2.1), libc6 (>= 2.33), libcap-ng0 (>= 0.7.9), libgcc-s1 (>= 3.0), libglib2.0-0 (>= 2.26.0), libseccomp2 (>= 0.0.0~20120605), libstdc++6 (>= 9), libusbguard0 (>= 1.0.0+ds), dbus
Homepage: https://usbguard.github.io/
Download-Size: 140 kB
APT-Manual-Installed: yes
APT-Sources: http://us.archive.ubuntu.com/ubuntu impish/universe amd64 Packages
Description: USB device authorization policy framework
 The USBGuard software framework helps to protect your computer against rogue
 USB devices (a.k.a. BadUSB) by implementing basic allow- and blocklisting
 capabilities based on device attributes.
```

* kernel versoin: `Linux biggray 5.15.5-76051505-generic #202111250933~1638201579~21.10~09f1aa7 SMP Mon Nov 29 16:23:13 U x86_64 x86_64 x86_64 GNU/Linux`

Cropi, 2022-01-03:
May I ask you to specify which rule causes the problem? If there are some problems with generating default rules, I will have to reproduce it first.

hartwork, 2022-02-05:
@bonedaddy any news?
It would be great to either have the output of `sudo usbguard-rule-parser -t -f /etc/usbguard/rules.conf` or a minimal reproducable example file that demos the error with latest Git `master`.