USBGuard/usbguard #500: Ubuntu install instructions (for 21.10)

pcraciunoiu, 2021-12-28:
Thought I'd leave these here in case others want to install, as it was a multi-step process with plenty of digging for me:

```

sudo apt install protobuf-compiler autoconf libtool libseccomp-dev libaudit-dev libcap-ng-dev libumockdev-dev libdbus-glib-1-dev libxml2-utils xsltproc libpolkit-gobject-1-dev 
./autogen.sh
./configure --with-crypto-library=gcrypt --with-bundled-catch --with-bundled-pegtl --enable-systemd --with-dbus
make
sudo make install
```

After these steps I still wasn't able to run `usbguard` from the `/usr/local/bin` path though, only from the build folder. Issue https://github.com/USBGuard/usbguard/issues/485

thomasmodeneis, 2022-01-02:
my install on latest version ends with:
```
  CXX      src/Library/libusbguard_la-IPCClientPrivate.lo
In file included from src/Library/IPCClientPrivate.hpp:27,
                 from src/Library/IPCClientPrivate.cpp:24:
./src/Library/IPC/Policy.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory
   10 | #include <google/protobuf/port_def.inc>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [Makefile:1763: src/Library/libusbguard_la-IPCClientPrivate.lo] Error 1
make[2]: Leaving directory '/opt/security/usbguard'
make[1]: *** [Makefile:2713: all-recursive] Error 1
make[1]: Leaving directory '/opt/security/usbguard'
make: *** [Makefile:1045: all] Error 2
```
any ideas ?

pcraciunoiu, 2022-01-02:
@thomasmodeneis Try searching for "ubuntu no such file or directory google/protobuf/port_def.inc"

I found this which may be helpful - https://github.com/BVLC/caffe/issues/6774

> sudo apt install protobuf-compiler

Cropi, 2022-01-03:
At first look it seems like that you do not have development files installed for protobuf.

> my install on latest version ends with:
>
> ```
>   CXX      src/Library/libusbguard_la-IPCClientPrivate.lo
> In file included from src/Library/IPCClientPrivate.hpp:27,
>                  from [...]

hartwork, 2022-01-28:
I see that Ubuntu 18.04 and after come with packages for USBGuard. Is this about running a self-built copy — about compilation — or about installation or about a particular compile error (that may have been fixed by now., given the green CI today)?

For self-building, the CI is doing roughly the same job here:
 https://github.com/USBGuard/usbguard/blob/76509752f80610be6149229a38390895838b4d84/.github/workflows/build.yml#L85-L120 

What can be done more for this ticket today?

pcraciunoiu, 2022-01-28:
If I recall, for me, the out of the box package from apt doesn't work.

I'm using this right now but still have issues with the built version not fully working without the apt version, maybe if I try the CI steps you linked I'll figure out what I'm missing.

hartwork, 2022-01-28:
@pcraciunoiu please be sure to test them in isolation: The self-building one with the Ubuntu package removed or the package with any self-built installed files removed (e.g. if you installed those to `/usr/local`). Otherwise they may end up interfering with each other and we'll have a hard time figuring things out then.

pcraciunoiu, 2022-01-28:
@hartwork OK, I remember what the issue is now.

So, the apt package does work, but it's very outdated - 0.7.2, released in January 2018!

If I just run the build, I get the Connection refused issue. As noted here - https://github.com/USBGuard/usbguard/issues/485#issuecomment-1004950390

I tried following the steps you posted from the CI workflow, and I still get the Connection refused after I `sudo make install` and run `./usbguard` or the global `usbguard`

I haven't found a way to fix that, so what I end up doing is mixing the two, by installing the apt package and then building this over it to get a newer version. Other than devices sometimes seeming to be forgotten between restarts (even with the generate-policy and install policy steps), I haven't had any issues.

I'd love to figure out the Connection refused problem, but I'm not sure how to approach it.

hartwork, 2022-01-28:
> So, the apt package does work, but it's very outdated - 0.7.2, released in January 2018!

@pcraciunoiu USBGuard is at version [1.0.0+ds-2](https://packages.ubuntu.com/hirsute/usbguard) in Ubuntu 21.04. The 0.7.2 you mention is Ubuntu 18.04. If you are using Ubuntu 18.04 that means you only get (some) security updates and are otherwise running software from 2018, four yours ago, by design. So it's no surprise that USBGuard is out of date there. If you'd like to be more up to date in general, an upgrade to e.g. Ubuntu 21.04 would help.

> If I just run the build, I get the Connection refused issue. As noted here - [#485 (comment)](https://github.com/USBGuard/usbguard/issues/485#issuecomment-1004950390)
>
> I tried following the steps you posted from the CI workflow, and I still get the Connection refused after I `sudo make install` and run `./usbguard` or the global `usbguard`
> 
> I haven't found a way to fix that, so what I end up doing is mixing the two, by installing the apt package and then building this over it to get a newer version. Other than devices sometimes seeming to be forgotten between restarts (even with the generate-policy and install policy steps), I haven't had any issues.
> 
> I'd love to figure out the Connection refused problem, but I'm not sure how to approach it.

There are two tickets about the "Connection refused" problem:

- #161
- #457

Do you have a file `/etc/usbguard/rules.conf` and its non-empty? If you don't, a command like `sudo sh -c "usbguard generate-policy > /etc/usbguard/rules.conf"` may be needed.
Please reconsider mixing the two: It's the worst of all worlds and it will be near impossible for other people to understand what's really happening on your system when you run into issues like these.

pcraciunoiu, 2022-01-28:
@hartwork maybe the debian repo was updated recently, or I misremember. I'm on Ubuntu 21.10 and I do get usbguard 1.0.0 now. So I'm going to give up on this build for now and work with the package version since that seems to be fine, and I won't be mixing the two either so everyone is happy. I ran a `make uninstall` and reinstalled the package with apt.

Yes I do have the file `/etc/usbguard/rules.conf` and still got the Connection refused error with the build.

hartwork, 2022-01-28:
> @hartwork maybe the debian repo was updated recently, or I misremember. I'm on Ubuntu 21.10 and I do get usbguard 1.0.0 now. So I'm going to give up on this build for now and work with the package [...]

@pcraciunoiu so 21.10 and package only, good, thank you.

> Yes I do have the file `/etc/usbguard/rules.conf` and still got the Connection refused error with the build.

Okay, so the USBGuard CLI cannot connect to the daemon I suppose? What's the precise command you're running that gets you that error? What are systemctl and journalctl saying about the status and log of the USBGuard daemon? Are there errors? Does it need restarting?

hartwork, 2022-01-28:
@pcraciunoiu thanks for the update, glad to hear things are working for you now, cool!
Regarding the packages need to install and all, there is a _distro-agnostic_ version at https://github.com/USBGuard/usbguard#compilation--installation . The distro-specific package names vary, e.g. on Alpine Linux the list is quite different from the list for Ubuntu. Making a list for each distro and keeping them up to date doesn't scale too well, so I'm not sure that's the best way to go forward, personally.

PR #523 (2022-02-04): Improve Readme (Ubuntu 21.10 instructions, help with first start, more) + make CI cover Ubuntu 21.10 (fixes #474, fixes #500)
Fixes #474 — Installation procedure and doc could be better
Fixes #500 — Ubuntu install instructions (for 21.10)
closed 2022-02-04 by radosroka