USBGuard/usbguard #501: KeyValueParser: Parser: Separator not found: syntax error

Bogdan107, 2022-01-07:
### Error with _usbguard 1.0.0-r5_:
```
/usr/sbin/usbguard-daemon -k -c /etc/usbguard/usbguard-daemon.conf
[1641564550.954] (E) KeyValueParser: Parser: Separator not found: syntax error
```

#### /etc/usbguard/usbguard-daemon.conf
```
$ grep -v -e "^#" -e "^$" /etc/usbguard/usbguard-daemon.conf
RuleFile=/etc/usbguard/rules.conf
ImplicitPolicyTarget=block
PresentDevicePolicy=apply-policy
PresentControllerPolicy=keep
InsertedDevicePolicy=apply-policy
AuthorizedDefault=none
RestoreControllerDeviceState=false
DeviceManagerBackend=uevent
IPCAllowedUsers=root
IPCAllowedGroups=
IPCAccessControlFiles=/etc/usbguard/IPCAccessControl.d/
DeviceRulesWithPort=false
AuditBackend=FileAudit
AuditFilePath=/var/lib/log/usbguard/usbguard-audit.log
HidePII=false
```

hartwork, 2022-02-05:
Hi @Bogdan107, I copied that comment-stripped version of yours into a file `issue501.conf` and the ran `usbguard-daemon` from Git `master`, and I get no parse errors:

```console
# ./usbguard-daemon -P -c issue501.conf -k -d |& fgrep Parsed:
[1644078201.317] (D) ConfigFilePrivate.cpp@155/parse: Parsed: RuleFile=/etc/usbguard/rules.conf
[1644078201.318] (D) ConfigFilePrivate.cpp@155/parse: Parsed: ImplicitPolicyTarget=block
[1644078201.318] (D) ConfigFilePrivate.cpp@155/parse: Parsed: PresentDevicePolicy=apply-policy
[1644078201.318] (D) ConfigFilePrivate.cpp@155/parse: Parsed: PresentControllerPolicy=keep
[1644078201.318] (D) ConfigFilePrivate.cpp@155/parse: Parsed: InsertedDevicePolicy=apply-policy
[1644078201.318] (D) ConfigFilePrivate.cpp@155/parse: Parsed: AuthorizedDefault=none
[1644078201.318] (D) ConfigFilePrivate.cpp@155/parse: Parsed: RestoreControllerDeviceState=false
[1644078201.318] (D) ConfigFilePrivate.cpp@155/parse: Parsed: DeviceManagerBackend=uevent
[1644078201.318] (D) ConfigFilePrivate.cpp@155/parse: Parsed: IPCAllowedUsers=root
[1644078201.318] (D) ConfigFilePrivate.cpp@155/parse: Parsed: IPCAllowedGroups=
[1644078201.318] (D) ConfigFilePrivate.cpp@155/parse: Parsed: IPCAccessControlFiles=/etc/usbguard/IPCAccessControl.d/
[1644078201.318] (D) ConfigFilePrivate.cpp@155/parse: Parsed: DeviceRulesWithPort=false
[1644078201.318] (D) ConfigFilePrivate.cpp@155/parse: Parsed: AuditBackend=FileAudit
[1644078201.318] (D) ConfigFilePrivate.cpp@155/parse: Parsed: AuditFilePath=/var/lib/log/usbguard/usbguard-audit.log
[1644078201.318] (D) ConfigFilePrivate.cpp@155/parse: Parsed: HidePII=false
```

If there is a problem, it must be in the comments then? Please help me reproduce the issue for Git `master`. Thank you!

hartwork, 2022-02-19:
@Bogdan107 any news?

Aleksanaa, 2022-05-08:
I suddenly got the same error on archlinux(after changing the kernel to linux-clear,but maybe it occured a long time ago but I just happened to check the system journal)
Haven't modified anything.The /etc/usbguard/usbguard-daemon.conf is actually the same as above.
usbguard-1.1.1-2

sbenner45, 2022-10-25:
Hello all,
I ran into this issue and it looks to be an issue with the way it is parsing. 

According to line 100 of KeyValueParserPrivate.cpp it appears the code assumes a comment does not start with a space and does not strip leading spaces either. Therefor, it parses commented lines that start with spaces as useful data.

```c++
      if ((line.size() < 1) || (line[0] == '#')) {
```

https://github.com/USBGuard/usbguard/blob/master/src/Library/KeyValueParserPrivate.cpp

To fix right now, strip your leading spaces of your commented lines so they start with the '#' character.

Example
```bash
sed -e 's/^[ \t]*#/#/' file.conf
```

skylinrcr01, 2022-10-25:
^ That fix worked for me!