USBGuard/usbguard #503: An option to `allow-device -p` regardless of the parent?

l29ah, 2022-01-21:
Usually when people want to allow a device connecting to the system, they would like to allow it connecting at whatever path (as they plug it in different ports or sometimes use hubs, docks, etc). Today they have to edit the rules file manually to remove the `parent-hash` after adding the device. It would be nice if there was a way to not include it in the rules file in the first place.

radosroka, 2022-01-27:
Yeah, PRs are welcomed :)

hartwork, 2022-02-02:
I'm wondering if it's this `true` that needs flipping to `false`:
https://github.com/USBGuard/usbguard/blob/64f7169ed346ed61fa5f25dafd897db4fa746ea0/src/Daemon/Daemon.cpp#L854
Would need a closer look.

radosroka, 2022-02-02:
IIRC generate-policy -X does something like that. We need to introduce similar flags for the command or invent some filter mechanism without changing the default behavior.