USBGuard/usbguard #504: Plans for a release >1.0.0?

hartwork, 2022-01-26:
Hi!

The [latest release 1.0.0](https://github.com/USBGuard/usbguard/releases/tag/usbguard-1.0.0) is one year old now and there are [28 unreleased commits with fixes after that on `master`](https://github.com/USBGuard/usbguard/compare/usbguard-1.0.0...master). Are there plans for a new release? Unblocking [PEGTL 3.2.x](https://github.com/USBGuard/usbguard/pull/453) for the package manager is one reason why I'm asking.

Thanks and best, Sebastian

radosroka, 2022-01-27:
I need to check all the PRs if there is something worth to include in another release. Do you need it anytime soon?

hartwork, 2022-01-27:
Hi @radosroka, not super super urgent, but it would get the fixes to users and unblock installation of PEGTL 3.2.x. I can try skip read through the 91 open issues if I find things that seem more critical than apparent today.

Issue #394 turned out to be a linking issue that potentially many distros (not just CentOS) may need to patch. Happy to discuss more in the ticket (to stay on topic here but your call).

hartwork, 2022-02-22:
For the record, as of today I think two tiny things would be good before a release and then we may be good to go for a new release in my eyes:
- merge #536 
- implement #537

CC @anudeepthika

closed 2022-03-21 by hartwork