USBGuard/usbguard #507: Should I bring back Astyle format enforcing to the CI?

hartwork, 2022-01-26:
I noticed the CI has some bit installing code formatter Astyle but then it doesn't use it and the current code base has also diverged from Astyle's style at a few places. Is it in your interest if I repair that and make the CI reject mis-formatted code again? It's not hard to do, but if you're happy that code style is *not* enforced right now, I would create a different pull request ripping the Astyle remains out. What do you think?

Best, Sebastian

radosroka, 2022-01-27:
Hello,

feel free to do it. It did work with travis some time ago but since we use github actions it's broken.

hartwork, 2022-01-27:
@radosroka I checked 83c191796df256eb48da09d9011b371ea163c80d but I don't see Travis enforcing Astyle formatting. Can you help me find it? It's not needed to go forward, but I'd be curious to see and compare.

radosroka, 2022-01-28:
The test itself is in https://github.com/USBGuard/usbguard/blob/master/src/Tests/Source/CheckScripts/code-style.sh. 

There is also script helper https://github.com/USBGuard/usbguard/blob/master/scripts/reformat-sources.sh that was used to format sources before creating PR.

hartwork, 2022-01-28:
> The test itself is in https://github.com/USBGuard/usbguard/blob/master/src/Tests/Source/CheckScripts/code-style.sh.

Found it and made #509 to fix it, yes. Have you seen it?

> There is also script helper https://github.com/USBGuard/usbguard/blob/master/scripts/reformat-sources.sh that was used to format sources before creating PR.

Interesting, hadn't seen that one before. It works to this day. It arguably formats some files too many that are not versioned in Git, but it's a minor glitch and not critical:
```console
# scripts/reformat-sources.sh
[..]
Formatted  scripts/..//src/Library/IPC/Policy.pb.h
Formatted  scripts/..//src/Library/IPC/Parameter.pb.h
Formatted  scripts/..//src/Library/IPC/Devices.pb.h
Formatted  scripts/..//src/Library/IPC/Rule.pb.h
Formatted  scripts/..//src/Library/IPC/Message.pb.h
Formatted  scripts/..//src/Library/IPC/Exception.pb.h
Formatted  scripts/..//src/build-config.h
```

radosroka, 2022-01-31:
> > The test itself is in https://github.com/USBGuard/usbguard/blob/master/src/Tests/Source/CheckScripts/code-style.sh.
> 
> Found it and made PR #509 to fix it, yes. Have you seen it?
> 
> > There is also script helper https://github.com/USBGuard/usbguard/blob/master/scripts/reformat-sources.sh that was used to format sources before creating PR.
> 
> Interesting, hadn't seen that one before. Script `reformat-sources.sh` in itself is not broken, the fixes in #509 make it work "for free" as a side effect. The script arguably formats some files too many (that are not versioned in Git), but it's a minor glitch and not critical. It's these files in practice:
> 
> ```
> # scripts/reformat-sources.sh
> [..]
> Formatted  scripts/..//src/Library/IPC/Policy.pb.h
> Formatted  scripts/..//src/Library/IPC/Parameter.pb.h
> Formatted  scripts/..//src/Library/IPC/Devices.pb.h
> Formatted  scripts/..//src/Library/IPC/Rule.pb.h
> Formatted  scripts/..//src/Library/IPC/Message.pb.h
> Formatted  scripts/..//src/Library/IPC/Exception.pb.h
> Formatted  scripts/..//src/build-config.h
> ```

We should exclude exclude them but it can be done later. I will check the PR.

hartwork, 2022-01-31:
> We should exclude exclude them but it can be done later.

@radosroka I have included that^^ with PR #509 now as well.

PR #509 (2022-02-01): Repair Astyle CI integration (fixes #507) + mass-fix formatting using Astyle
Fixes #507

My vote for keeping individual commits rather than squash-merging.

CC @radosroka
closed 2022-02-01 by radosroka