USBGuard/usbguard #524: Failed to start USBGuard daemon.

hail-mary, 2022-02-04:
OS: OpenSuse Tumbleweed
Kernel: 5.16.4-1
usbguard: 1.0.0-2.4

Description: After installation and configuration of usbguard-deamon.conf and rules.conf the service fails to start every time.

```
user@computer:/etc/usbguard> sudo systemctl start usbguard
Job for usbguard.service failed because the control process exited with error code.
See "systemctl status usbguard.service" and "journalctl -xeu usbguard.service" for details.
user@computer:/etc/usbguard> sudo systemctl status usbguard.service
× usbguard.service - USBGuard daemon
     Loaded: loaded (/usr/lib/systemd/system/usbguard.service; enabled; vendor preset: disable>
     Active: failed (Result: exit-code) since Fri 2022-02-04 17:11:45 CET; 13s ago
       Docs: man:usbguard-daemon(8)
    Process: 3929 ExecStart=/usr/sbin/usbguard-daemon -f -s -c /etc/usbguard/usbguard-daemon.c>
        CPU: 33ms

Feb 04 17:11:45 computer systemd[1]: usbguard.service: Scheduled restart job, restart counter>
Feb 04 17:11:45 computer systemd[1]: Stopped USBGuard daemon.
Feb 04 17:11:45 computer systemd[1]: usbguard.service: Start request repeated too quickly.
Feb 04 17:11:45 computer systemd[1]: usbguard.service: Failed with result 'exit-code'.
Feb 04 17:11:45 computer systemd[1]: Failed to start USBGuard daemon.
```

```
user@computer:/etc/usbguard> ls -al
total 12
drwxr-xr-x 1 root root   96 Feb  4 17:08 .
drwxr-xr-x 1 root root 4186 Feb  4 15:44 ..
drwxr-xr-x 1 root root    0 Jan 28 00:21 IPCAccessControl.d
-rw------- 1 root root 2505 Feb  4 16:21 rules.conf
-rw------- 1 root root 4985 Feb  4 17:05 usbguard-daemon.conf
```
```
user@computer:/etc/usbguard> sudo cat /etc/usbguard/usbguard-daemon.conf 
[sudo] password for root: 
#
# Rule set file path.
#
# The USBGuard daemon will use this file to load the policy
# rule set from it and to write new rules received via the
# IPC interface.
#
# RuleFile=/path/to/rules.conf
#
RuleFile=/etc/usbguard/rules.conf

#
# Implicit policy target.
#
# How to treat devices that don't match any rule in the
# policy. One of:
#
# * allow  - authorize the device
# * block  - block the device
# * reject - remove the device
#
ImplicitPolicyTarget=block

#
# Present device policy.
#
# How to treat devices that are already connected when the
# daemon starts. One of:
#
# * allow        - authorize every present device
# * block        - deauthorize every present device
# * reject       - remove every present device
# * keep         - just sync the internal state and leave it
# * apply-policy - evaluate the ruleset for every present
#                  device
#
PresentDevicePolicy=apply-policy

#
# Present controller policy.
#
# How to treat USB controllers that are already connected
# when the daemon starts. One of:
#
# * allow        - authorize every present device
# * block        - deauthorize every present device
# * reject       - remove every present device
# * keep         - just sync the internal state and leave it
# * apply-policy - evaluate the ruleset for every present
#                  device
#
PresentControllerPolicy=apply-policy

#
# Inserted device policy.
#
# How to treat USB devices that are already connected
# *after* the daemon starts. One of:
#
# * block        - deauthorize every present device
# * reject       - remove every present device
# * apply-policy - evaluate the ruleset for every present
#                  device
#
InsertedDevicePolicy=apply-policy

#
# Restore controller device state.
#
# The USBGuard daemon modifies some attributes of controller
# devices like the default authorization state of new child device
# instances. Using this setting, you can controll whether the
# daemon will try to restore the attribute values to the state
# before modificaton on shutdown.
#
# SECURITY CONSIDERATIONS: If set to true, the USB authorization
# policy could be bypassed by performing some sort of attack on the
# daemon (via a local exploit or via a USB device) to make it shutdown
# and restore to the operating-system default state (known to be permissive).
#
RestoreControllerDeviceState=false

#
# Device manager backend
#
# Which device manager backend implementation to use. One of:
#
# * uevent   - Netlink based implementation which uses sysfs to scan for present
#              devices and an uevent netlink socket for receiving USB device
#              related events.
# * umockdev - umockdev based device manager capable of simulating devices based
#              on umockdev-record files. Useful for testing.
#
DeviceManagerBackend=uevent

#!!! WARNING: It's good practice to set at least one of the !!!
#!!!          two options bellow. If none of them are set,  !!!
#!!!          the daemon will accept IPC connections from   !!!
#!!!          anyone, thus allowing anyone to modify the    !!!
#!!!          rule set and (de)authorize USB devices.       !!!

#
# Users allowed to use the IPC interface.
#
# A space delimited list of usernames that the daemon will
# accept IPC connections from.
#
# IPCAllowedUsers=username1 username2 ...
#
IPCAllowedUsers=user

#
# Groups allowed to use the IPC interface.
#
# A space delimited list of groupnames that the daemon will
# accept IPC connections from.
#
# IPCAllowedGroups=groupname1 groupname2 ...
#
#IPCAllowedGroups=

#
# IPC access control definition files path.
#
# The files at this location will be interpreted by the daemon
# as access control definition files. The (base)name of a file
# should be in the form:
#
#   [user][:<group>]
#
# and should contain lines in the form:
#
#   <section>=[privilege] ...
#
# This way each file defines who is able to connect to the IPC
# bus and what privileges he has.
#
IPCAccessControlFiles=/etc/usbguard/IPCAccessControl.d/

#
# Generate device specific rules including the "via-port"
# attribute.
#
# This option modifies the behavior of the allowDevice
# action. When instructed to generate a permanent rule,
# the action can generate a port specific rule. Because
# some systems have unstable port numbering, the generated
# rule might not match the device after rebooting the system.
#
# If set to false, the generated rule will still contain
# the "parent-hash" attribute which also defines an association
# to the parent device. See usbguard-rules.conf(5) for more
# details.
#
DeviceRulesWithPort=false

#
# USBGuard Audit events log backend
#
# One of:
#
# * FileAudit - Log audit events into a file specified by
#               AuditFilePath setting (see below)
# * LinuxAudit - Log audit events using the Linux Audit
#                subsystem (using audit_log_user_message)
#
AuditBackend=FileAudit

#
# USBGuard audit events log file path.
#
AuditFilePath=/var/log/usbguard/usbguard-audit.log
```

hartwork, 2022-02-04:
Hi @hail-mary , could you share `journalctl -xeu usbguard.service` output as well? I think first verbatim above is only `systemctl status usbguard.service` output, correct?

hartwork, 2022-02-04:
PS: I understand that `rules.conf` content may be sensitive. Could you check if it's non empty though and elaborate how you generated that file?

hail-mary, 2022-02-04:
Thank you for the quick response. Here is the output from journalctl.
```
user@computer:~> sudo journalctl -xeu usbguard.service
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ Automatic restarting of the unit usbguard.service has been scheduled, as the result for
░░ the configured Restart= setting for the unit.
Feb 04 20:56:32 computer systemd[1]: Stopped USBGuard daemon.
░░ Subject: A stop job for unit usbguard.service has finished
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ A stop job for unit usbguard.service has finished.
░░ 
░░ The job identifier is 6978 and the job result is done.
Feb 04 20:56:32 computer systemd[1]: usbguard.service: Start request repeated too quickly.
Feb 04 20:56:32 computer systemd[1]: usbguard.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ The unit usbguard.service has entered the 'failed' state with result 'exit-code'.
Feb 04 20:56:32 computer systemd[1]: Failed to start USBGuard daemon.
░░ Subject: A start job for unit usbguard.service has failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ A start job for unit usbguard.service has finished with a failure.
░░ 
░░ The job identifier is 6978 and the job result is failed.
``` 
The rules.conf was generated by the following command, does contain rules and does not contain comments.
``` 
usbguard generate-policy | sudo tee -a /etc/usbguard/rules.conf
```

hartwork, 2022-02-04:
@hail-mary thanks for the info. Rule generation looks good. The journctl output doesn't really show any messages from USBGuard itself :thinking: I'm not sure why. Could you try manual `sudo usbguard-daemon -k -d` and see if it explains its problem? The commend will stay in the foreground, log to the console (`-k`), include debug output (`-d`), and can be terminated with plain <kbd>Ctrl</kbd>+<kbd>C</kbd>. Does it say anything helpful?

hail-mary, 2022-02-04:
``` 
user@computer:~> sudo usbguard-daemon -k -d
[sudo] password for root: 
[1644005630.211] (i) NSHandler Loading...
[1644005630.212] (i) separator -> :
[1644005630.212] (i) keys:
[1644005630.212] (i) --->usbguard
[1644005630.212] (i) NSHandler Loaded
[1644005630.212] (i) Loading configuration from /etc/usbguard/usbguard-daemon.conf
[1644005630.212] (i) File has correct permissions.
[1644005630.212] (D) ConfigFilePrivate.cpp@155/parse: Parsed: RuleFile=/etc/usbguard/rules.conf
[1644005630.212] (D) ConfigFilePrivate.cpp@155/parse: Parsed: ImplicitPolicyTarget=block
[1644005630.212] (D) ConfigFilePrivate.cpp@155/parse: Parsed: PresentDevicePolicy=apply-policy
[1644005630.212] (D) ConfigFilePrivate.cpp@155/parse: Parsed: PresentControllerPolicy=apply-policy
[1644005630.212] (D) ConfigFilePrivate.cpp@155/parse: Parsed: InsertedDevicePolicy=apply-policy
[1644005630.212] (D) ConfigFilePrivate.cpp@155/parse: Parsed: RestoreControllerDeviceState=false
[1644005630.212] (D) ConfigFilePrivate.cpp@155/parse: Parsed: DeviceManagerBackend=uevent
[1644005630.212] (D) ConfigFilePrivate.cpp@155/parse: Parsed: IPCAllowedUsers=user
[1644005630.212] (D) ConfigFilePrivate.cpp@155/parse: Parsed: IPCAccessControlFiles=/etc/usbguard/IPCAccessControl.d/
[1644005630.212] (D) ConfigFilePrivate.cpp@155/parse: Parsed: DeviceRulesWithPort=false
[1644005630.212] (D) ConfigFilePrivate.cpp@155/parse: Parsed: AuditBackend=FileAudit
[1644005630.212] (D) ConfigFilePrivate.cpp@155/parse: Parsed: AuditFilePath=/var/log/usbguard/usbguard-audit.log
[1644005630.212] (i) Loading NSSwitch...
[1644005630.212] (i) Loading nsswitch from /etc/nsswitch.conf
[1644005630.212] (E) NSSwitch parsing: /etc/nsswitch.conf: No such file or directory
```

hartwork, 2022-02-04:
> ```
> user@computer:~> sudo usbguard-daemon -k -d
> [..]
> [1644005630.212] (i) Loading NSSwitch...
> [1644005630.212] (i) Loading nsswitch from /etc/nsswitch.conf
> [1644005630.212] (E) NSSwitch parsing: /etc/nsswitch.conf: No such file or directory
> ```

@hail-mary that^^ is key. Could it be that your libc is not glibc but something else like musl? On my local Gentoo box file `/etc/nsswitch.conf` is owned by package glibc. I also found related issue #423 where it seems the agreed fix was to patch away the related code in Alpine Linux packaging with a patch like https://git.alpinelinux.org/aports/tree/testing/usbguard/no-nsswitch-conf.patch?id=107d2477c81ea3b0b138d6c87fcb887c296d747d that Alpine seems to apply still today. I'm not sure that's something I would call a solution personally. For a quick workaround, there is a hint at https://github.com/USBGuard/usbguard/issues/423#issuecomment-706721300 that an empty file a la `sudo touch /etc/nsswitch.conf` may work. I'd be curious to hear more about your libc situation on SUSE.

hail-mary, 2022-02-04:
glibc is installed and up-to-date.
```
user@computer:~> sudo zypper info glibc
[sudo] password for root: 
Loading repository data...
Reading installed packages...

Information for package glibc:
------------------------------
Repository     : Main Repository (OSS)
Name           : glibc
Version        : 2.34-4.3
Arch           : x86_64
Vendor         : openSUSE
Installed Size : 6,1 MiB
Installed      : Yes
Status         : up-to-date
Source package : glibc-2.34-4.3.src
Summary        : Standard Shared Libraries (from the GNU C Library)
Description    : 
    The GNU C Library provides the most important standard libraries used
    by nearly all programs: the standard C library, the standard math
    library, and the POSIX thread library. A system is not functional
    without these libraries.
```

hartwork, 2022-02-04:
> OK I think I remember what it is. [OpenSUSE is going a step beyond what most distributions do, to be able to provide transactional server functionality.](https://en.opensuse.org/openSUSE:Packaging_UsrEtc). It creates files only in /usr/etc until the package was customized.
> 
> A simple copy fixed the issue! Thank you for your assistance!
> 
> ```
> sudo cp /usr/etc/nsswitch.conf /etc/
> ```

I see. It matches what I see on a Docker Tumbleweed shell:

```console
# lsb_release -a
LSB Version:    n/a
Distributor ID: openSUSE
Description:    openSUSE Tumbleweed
Release:        20220124
Codename:       n/a

# rpm -qf /etc/nsswitch.conf
glibc-2.34-4.3.x86_64

# rpm -ql glibc | fgrep nsswitch
/etc/nsswitch.conf
/usr/etc/nsswitch.conf

# ls -l /etc/nsswitch.conf /usr/etc/nsswitch.conf
ls: cannot access '/etc/nsswitch.conf': No such file or directory
-rw-r--r-- 1 root root 2190 Jan  7 13:01 /usr/etc/nsswitch.conf
```

@hail-mary should the issue be closed?

hail-mary, 2022-02-04:
The issue should be closed. It is up to the SUSE developers to properly update the package. Thank you for your help.

hartwork, 2022-02-04:
@hail-mary okay cool. I don't have permissions to close tickets here. If you see a pushable close button, could you push it for me?

closed 2022-02-04 by hail-mary