USBGuard/usbguard #532: What would be the reason for a hash to change?

duck-rh, 2022-02-14:
Quack,

I'm trying to understand what happened on my laptop and I'm not implying usbguard is doing anything wrong but since the device hash is, IIUC, computed by usbguard I thought you might know more about this. Also I would have used the mailing-list for this kind of question but it looks kind of deserted.

Today I raised my laptop (T490s + dock) from sleep as usual and could not get any IP, external monitor was not working, and I could not find any lead in the logs. Replugging my dock did not help and rebooting led to a BIOS message about the PSU having not enough wattage (charging goes through USB too). But then I noticed usbguard telling me some device was blocked and I was intrigued. It happened the hash for my dock (plugged though USB C) had changed and the parent USB hub as well. I added an allow rule, restarted everything and all is working fine again.

Kernel updates might have been installed but since I did not reboot that should not affect anything. fwupd had an empty history and anyway was missing a component to actually being able to push changes. Any idea what could have triggered the changes?

Regards.
\\_o<

duck-rh, 2022-02-14:
@hartwork thanks for clarifying. It makes sense but I don't see how these could change in between going to sleep. The `name` and `serial` have not changed for sure. I do not have the other past information because after rebooting several times to check everything and try to reproduce I just noticed the parent-hash is now back to its original self (the one matching the old rule, but I made the new rule more lax and it contains less info).

Looking at what [the kernel does](https://www.kernel.org/doc/html/v4.13/driver-api/usb/persist.html) it seems that suspend is more risky than I would have imagined. Maybe this annoying USB cable connecting to the dock could have moved and partially disconnected and caused this situation (it happens from time to time but never caused such a mess, so this is a wild guess).

hartwork, 2022-02-14:
Maybe not a strong argument, but: it could be a bit flip in hardware. I should re-watch [DEFCON 19: Bit-squatting: DNS Hijacking Without Exploitation (w speaker)](https://www.youtube.com/watch?v=aT7mnSstKGs). I remember the bottom line was: Bitflips are more likely than I thought they were before watching the talk.

duck-rh, 2022-02-28:
Alright, thanks @hartwork for the suggestion. If noone else has experienced this issue and I myself used this hardware the same way for years without such problem, hopefully this won't happen again and there's no reason to bother people longer.

closed 2022-02-28 by duck-rh