USBGuard/usbguard #537: configuration fails without polkit

anudeepthika, 2022-02-16:
Configuring usbguard without polkit is failing with the below error.

`configure: error: Required D-Bus modules (dbus-1, gio-2.0) not found!`

The above error is encountered due to the commit 2c8c1ae10a6330736bf2a2d7d8682674b1db47dc.
Can someone help me with a solution to skip this error while configuring without polkit?

anudeepthika, 2022-02-21:
Hi @hartwork,

Thank you for your response.
Sorry for my delay in replying back. I was analyzing the polkit impact on our environment.

In an Embedded System, different constraints with regards to policy enforcement would be implemented, and having polkit in our environment is throwing conflicts.

Could you please provide "--without-polkit" option so that polkit can be disabled depending on the use case and still remain enabled by default?

hartwork, 2022-02-21:
@anudeepthika that helps, but I still have questions, to be sure I understand the full picture and our options:
- Do you need `--with-dbus` and USBGuard DBus functionality enabled in that environment?
- Do you have a Polkit daemon running in that environment?
- Is adjusting the Polkit policy file to e.g. allow use of USBGuard D-Bus methods to any user on the system without interactive authorization an option? (That would require no changes in Git.)

anudeepthika, 2022-02-22:
Hi @hartwork,

There is no polkit daemon running in our environment and enabling it is making a huge impact. I request `--with-dbus` and USBGuard DBus functionality enabled without polkit.

hartwork, 2022-02-22:
Hi @anudeepthika ,

I guess then maybe the best way forward would be to introduce a new configure option `--with(out)-dbus-security` that defaults to enabled and have that control a C++ preprocessor macro that makes `DBusBridge::isAuthorizedByPolkit` always return `true` and print a warning when disabled, and do it's regular current job when enabled.

Best, Sebastian

hartwork, 2022-02-22:
PS: The idea with `--without-dbus-security` naming is that making that part of `--without-polkit` would do nothing to warn the user that operation with D-Bus but without talking to Polkit is not something secure for e.g. a regular desktop environment. I consider that essential.

hartwork, 2022-02-22:
> There is no polkit daemon running in our environment and enabling it, is making a huge impact.

@anudeepthika could you elaborate what kind of impact enabling the daemon would have?

radosroka, 2022-02-23:
Why do you need the dbus so bad? If you don't want polkit on your system you can just stay with usbguard CLI and IPC perms.