USBGuard/usbguard #539: Feature request

ppasserini, 2022-03-04:
I wonder if could be possible to add an option to deny/allow USB devices according to the user.

hartwork, 2022-03-04:
@ppasserini please elaborate on the idea more and what use case that that would help.

ppasserini, 2022-03-05:
@hartwork Thank you for your reply.
I'd like to deny the use of USB thumbdrive to some users, for istance guests. So I wonder if it's possible to allow all my user to attach usb devices without restriction (very easy and already possible with usbguard) but to deny this chance if the logged inuser is a guest user. 
Thanks

hartwork, 2022-03-05:
@ppasserini thanks for elaborating.

radosroka, 2022-03-10:
From Linux point of view it is impossible to pair the usb device with user or with a session. It is connected to the system and not to the user session. When there are two users logged in the system, the first one is normal and the second one is guest. The usb drive has been connected to the system. How would you distinguish who is the owner?

You can only set who can allow the devices. If the default policy is set to block and all of your users are able to manage these devices(via USBGuard CLI) except for guests. 

I believe this is what you wanted to achieve.

ppasserini, 2022-03-10:
@radosroka Thank you for your explaination.
My plan was to allow all my users to connect USB devices and, by default, block all guest users. According to your tip the only way to achieve the result is to allow only some USB device (thumb drive) and block everything else.

radosroka, 2022-03-11:
I'll close the issue.

closed 2022-03-11 by radosroka