USBGuard/usbguard #540: Ignoring access control file because of malformed name

brphilly, 2022-03-05:
Ever since upgrading to usbguard 1.1.0, I'm getting this error `ERROR: IPC connect: service=usbguard: Operation not permitted` when trying to use usbguard as my user. I ran `usbguard-daemon -d` and the logs showed
```
[1646452999.870] (D) Daemon.cpp@262/loadConfiguration: Setting IPCAllowedUsers to { root }
[1646452999.870] (T) Daemon.cpp@1086/addIPCAllowedUser: user=root
[1646452999.870] (D) Daemon.cpp@274/loadConfiguration: Setting IPCAllowedGroups to {  }
[1646452999.870] (D) Daemon.cpp@284/loadConfiguration: Setting DeviceRulesWithPort to false
[1646452999.870] (i) File has correct permissions.
[1646452999.870] (i) Loading IPC access control files at /etc/usbguard/IPCAccessControl.d
[1646452999.870] (T) Utility.cpp@361/loadFiles: L: brady : /etc/usbguard/IPCAccessControl.d/brady
[1646452999.870] (i) Loading IPC access control file /etc/usbguard/IPCAccessControl.d/brady
[1646452999.870] (W) Ignoring access control file because of malformed name: brady
```
The file `/etc/usbguard/IPCAccessControl.d/brady` was created by
```
sudo usbguard add-user brady --policy list --devices list,listen --exceptions listen --parameters list,listen
```
and it has the correct permissions.

I saw #479 and thought maybe that is causing the issue. I'm using Arch Linux in case it matters.

hartwork, 2022-03-05:
Hi @brphilly,

a quick look at the code reveals that the error message is produced at… https://github.com/USBGuard/usbguard/blob/87e5c2dac79ca6edbe500391436aa565b59304a1/src/Daemon/Daemon.cpp#L466-L472 … which means that function `parseIPCAccessControlFilename` threw an exception. The way I read function `parseIPCAccessControlFilename` at… https://github.com/USBGuard/usbguard/blob/87e5c2dac79ca6edbe500391436aa565b59304a1/src/Daemon/Daemon.cpp#L446-L457 … it seems to expect filenames containing a colon to separator a user name from a group name. Your filename has no colon. Note that this code was introduce with commit 67b811927275b8ce0f1863d9a9efd33019789211 which was first shipped in release 0.7.0. So it seems you need a different filename. Does that make sense?

Best, Sebastian

hartwork, 2022-03-05:
@brphilly any chance you could try if pull request #541 fully fixes this problem for you?

brphilly, 2022-03-05:
> @brphilly any chance you could try if pull request #541 fully fixes this problem for you?

Just compiled this PR and can confirm it fixed the problem. The logs now show
```
[1646502018.264] (i) Loading IPC access control files at /etc/usbguard/IPCAccessControl.d
[1646502018.264] (T) Utility.cpp@361/loadFiles: L: brady : /etc/usbguard/IPCAccessControl.d/brady
[1646502018.264] (i) Loading IPC access control file /etc/usbguard/IPCAccessControl.d/brady
[1646502018.264] (T) Daemon.cpp@1090/addIPCAllowedUser: user=brady
```
and everything seems to work. Thanks so much for the quick fix!

hartwork, 2022-03-05:
@brphilly thanks for testing and reporting back! :+1: 

@radosroka any chance for a release 1.1.1 with the regression fix from PR #541 ?

hartwork, 2022-03-15:
> @radosroka any chance for a release 1.1.1 with the regression fix from PR #541 ?

@radosroka I'm currently considering to backport PR #541 in Gentoo packaging of USBGuard but since it's a regression fix from 1.0.0, all distros should ideally have this patch and not need to accumulate distro-agnostic patches. Any chance you could cut release 1.1.1 from Git `master`?

radosroka, 2022-03-15:
> @radosroka I'm currently considering to backport PR #541 in Gentoo packaging of USBGuard but since it's a regression fix from 1.0.0, all distros should ideally have this patch and not need to [...]

I'll do it later today.