USBGuard/usbguard #543: device not properly authorized after docked suspend/resume

khimaros, 2022-03-24:
**description**

when external monitor/dock which is disconnected (causing device to suspend) and then reattached (causing device to resume), there appears to be a race condition which causes the device not to be correctly authorized by usbguard.

**reproduction steps**

1. configure laptop to suspend when lid closed and on battery power
1. attach laptop to external monitor/dock via USB-C with power delivery
1. close the laptop lid while attached and begin using external display only
1. unplug the USB-C cable from laptop (laptop should suspend)
1. reattach the USB-C cable and resume the laptop
1. unlock the screen (gnome in my case)

**what happened?**

looking at `usbguard list-devices` most devices correctly get `allow` permission

however, some devices (eg. external sound card) connected via the dock are still in `block`

upon running `systemctl restart usbguard`, all devices correctly are `allow`

**what should happen?**

all devices should have the same state on resume and not require restarting the usbguard service.

**additional detail**

on suspend/resume, devices are processed in a slightly different order vs. unplug/plug.

this may be a "feature" of the Framework laptop which has modular ports.

in the suspend/resume case, the USB 3.0 hub (device 2.1) is processed before the sound card.

**system logs**

keep lid open and unplug/replug (works fine):

```
[282866.662182] usb 2-1: New USB device found, idVendor=0bda, idProduct=0411, bcdDevice= 1.23
[282866.662188] usb 2-1: Product: 4-Port USB 3.0 Hub
[282866.662189] usb 2-1: Manufacturer: Generic
[282866.662269] usb 2-1: Device is not authorized for usage
[282866.678162] usb 2-1: authorized to connect

[282866.801352] usb 3-1: New USB device found, idVendor=0bda, idProduct=5411, bcdDevice= 1.23
[282866.801362] usb 3-1: Product: 4-Port USB 2.0 Hub
[282866.801363] usb 3-1: Manufacturer: Generic
[282866.801693] usb 3-1: Device is not authorized for usage
[282866.808126] usb 3-1: authorized to connect

[282867.410552] usb 3-1.2: New USB device found, idVendor=1235, idProduct=8210, bcdDevice= 5.ff
[282867.410565] usb 3-1.2: Product: Scarlett 2i2 USB
[282867.410567] usb 3-1.2: Manufacturer: Focusrite
[282867.410818] usb 3-1.2: Device is not authorized for usage
[282867.667983] usb 3-1.2: Focusrite Scarlett Gen 2/3 Mixer Driver enabled pid=0x8210
[282867.670281] usb 3-1.2: authorized to connect
```

```
Mar 23 16:18:07 usbguard-daemon[180484]: uid=0 pid=180479 result='SUCCESS' device.rule='block id 0bda:5411 serial "" name "4-Port USB 2.0 Hub" hash "0x2npO6BRSDC2xIDDgI/r4xfy6sMVtARM6Ei4PQWg2Q=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "3-1" with-interface { 09:00:01 09:00:02 } with-connect-type "hotplug"' device.system_name='/devices/pci0000:00/0000:00:14.0/usb3/3-1' type='Device.Insert'
Mar 23 16:18:07 usbguard-daemon[180484]: uid=0 pid=180479 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb3/3-1' target.new='allow' device.rule='block id 0bda:5411 serial "" name "4-Port USB 2.0 Hub" hash "0x2npO6BRSDC2xIDDgI/r4xfy6sMVtARM6Ei4PQWg2Q=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "3-1" with-interface { 09:00:01 09:00:02 } with-connect-type "hotplug"' target.old='block' type='Policy.Device.Update'
Mar 23 16:18:07 usbguard-daemon[180484]: Ignoring unknown UEvent action: sysfs_devpath=/devices/pci0000:00/0000:00:14.0/usb3/3-1 action=change
Mar 23 16:18:08 usbguard-daemon[180484]: uid=0 pid=180479 result='SUCCESS' device.rule='block id 1235:8210 serial "Y8AAPA60BA8E51" name "Scarlett 2i2 USB" hash "fmhmYN+t28mmrYQbKydXhabL0jc5yTVRdMlw3bJ+UmI=" parent-hash "0x2npO6BRSDC2xIDDgI/r4xfy6sMVtARM6Ei4PQWg2Q=" via-port "3-1.2" with-interface { 01:01:20 01:02:20 01:02:20 01:02:20 01:02:20 ff:01:20 } with-connect-type "unknown"' device.system_name='/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.2' type='Device.Insert'
Mar 23 16:18:08 usbguard-daemon[180484]: uid=0 pid=180479 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.2' target.new='allow' device.rule='block id 1235:8210 serial "Y8AAPA60BA8E51" name "Scarlett 2i2 USB" hash "fmhmYN+t28mmrYQbKydXhabL0jc5yTVRdMlw3bJ+UmI=" parent-hash "0x2npO6BRSDC2xIDDgI/r4xfy6sMVtARM6Ei4PQWg2Q=" via-port "3-1.2" with-interface { 01:01:20 01:02:20 01:02:20 01:02:20 01:02:20 ff:01:20 } with-connect-type "unknown"' target.old='block' type='Policy.Device.Update'
```

with suspend/resume (failure mode):

```
[283614.614420] usb 2-1: New USB device found, idVendor=0bda, idProduct=0411, bcdDevice= 1.23
[... 12 lines cut ...]
[283615.276761] usb 3-1.2: Product: Scarlett 2i2 USB
[283615.276763] usb 3-1.2: Manufacturer: Focusrite
[283615.277074] usb 3-1.2: Device is not authorized for usage
```

note above the device is never authorized to connect.

```
Mar 23 16:21:10 usbguard-daemon[180484]: uid=0 pid=180479 result='SUCCESS' device.rule='block id 0bda:5411 serial "" name "4-Port USB 2.0 Hub" hash "0x2npO6BRSDC2xIDDgI/r4xfy6sMVtARM6Ei4PQWg2Q=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "3-1" with-interface { 09:00:01 09:00:02 } with-connect-type "hotplug"' device.system_name='/devices/pci0000:00/0000:00:14.0/usb3/3-1' type='Device.Insert'
Mar 23 16:21:10 usbguard-daemon[180484]: uid=0 pid=180479 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb3/3-1' target.new='block' device.rule='block id 0bda:5411 serial "" name "4-Port USB 2.0 Hub" hash "0x2npO6BRSDC2xIDDgI/r4xfy6sMVtARM6Ei4PQWg2Q=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "3-1" with-interface { 09:00:01 09:00:02 } with-connect-type "hotplug"' target.old='block' type='Policy.Device.Update'
Mar 23 16:21:10 usbguard-daemon[180484]: Ignoring unknown UEvent action: sysfs_devpath=/devices/pci0000:00/0000:00:14.0/usb3/3-1 action=change
Mar 23 16:21:10 usbguard-daemon[180484]: uid=0 pid=180479 result='SUCCESS' device.rule='block id 0bda:0411 serial "" name "4-Port USB 3.0 Hub" hash "f+TIJlWiW3v9RkpyW/4cQQHZlS5jcpq3YxkacQEAiP4=" parent-hash "f+TIJlWiW3v9RkpyW/4cQQHZlS5jcpq3YxkacQEAiP4=" via-port "2-1.4" with-interface 09:00:00 with-connect-type "unknown"' device.system_name='/devices/pci0000:00/0000:00:0d.0/usb2/2-1/2-1.4' type='Device.Insert'
Mar 23 16:21:10 usbguard-daemon[180484]: uid=0 pid=180479 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:0d.0/usb2/2-1/2-1.4' target.new='block' device.rule='block id 0bda:0411 serial "" name "4-Port USB 3.0 Hub" hash "f+TIJlWiW3v9RkpyW/4cQQHZlS5jcpq3YxkacQEAiP4=" parent-hash "f+TIJlWiW3v9RkpyW/4cQQHZlS5jcpq3YxkacQEAiP4=" via-port "2-1.4" with-interface 09:00:00 with-connect-type "unknown"' target.old='block' type='Policy.Device.Update'
Mar 23 16:21:10 usbguard-daemon[180484]: Ignoring unknown UEvent action: sysfs_devpath=/devices/pci0000:00/0000:00:0d.0/usb2/2-1/2-1.4 action=change
Mar 23 16:21:10 usbguard-daemon[180484]: uid=0 pid=180479 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:0d.0/usb2/2-1/2-1.4' target.new='allow' device.rule='block id 0bda:0411 serial "" name "4-Port USB 3.0 Hub" hash "f+TIJlWiW3v9RkpyW/4cQQHZlS5jcpq3YxkacQEAiP4=" parent-hash "f+TIJlWiW3v9RkpyW/4cQQHZlS5jcpq3YxkacQEAiP4=" via-port "2-1.4" with-interface 09:00:00 with-connect-type "unknown"' target.old='block' type='Policy.Device.Update'
Mar 23 16:21:10 usbguard-daemon[180484]: uid=0 pid=180479 result='SUCCESS' device.rule='block id 1235:8210 serial "Y8AAPA60BA8E51" name "Scarlett 2i2 USB" hash "fmhmYN+t28mmrYQbKydXhabL0jc5yTVRdMlw3bJ+UmI=" parent-hash "0x2npO6BRSDC2xIDDgI/r4xfy6sMVtARM6Ei4PQWg2Q=" via-port "3-1.2" with-interface { 01:01:20 01:02:20 01:02:20 01:02:20 01:02:20 ff:01:20 } with-connect-type "unknown"' device.system_name='/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.2' type='Device.Insert'
Mar 23 16:21:10 usbguard-daemon[180484]: uid=0 pid=180479 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.2' target.new='block' device.rule='block id 1235:8210 serial "Y8AAPA60BA8E51" name "Scarlett 2i2 USB" hash "fmhmYN+t28mmrYQbKydXhabL0jc5yTVRdMlw3bJ+UmI=" parent-hash "0x2npO6BRSDC2xIDDgI/r4xfy6sMVtARM6Ei4PQWg2Q=" via-port "3-1.2" with-interface { 01:01:20 01:02:20 01:02:20 01:02:20 01:02:20 ff:01:20 } with-connect-type "unknown"' target.old='block' type='Policy.Device.Update'
```

note the missing `target.new='allow'` in this case.

```
61: allow id 0bda:5411 serial "" name "4-Port USB 2.0 Hub" hash "0x2npO6BRSDC2xIDDgI/r4xfy6sMVtARM6Ei4PQWg2Q=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "3-1" with-interface { 09:00:01 09:00:02 } with-connect-type "hotplug"
63: block id 1235:8210 serial "Y8AAPA60BA8E51" name "Scarlett 2i2 USB" hash "fmhmYN+t28mmrYQbKydXhabL0jc5yTVRdMlw3bJ+UmI=" parent-hash "0x2npO6BRSDC2xIDDgI/r4xfy6sMVtARM6Ei4PQWg2Q=" via-port "3-1.2" with-interface { 01:01:20 01:02:20 01:02:20 01:02:20 01:02:20 ff:01:20 } with-connect-type "unknown"
```

khimaros, 2022-03-26:
another hint: when the device ends up in this state, i can disconnect/reattach *only* the sound card and it will be properly authorized by usbguard. it's possible the race condition involves the initialization of the device itself after receiving USB power.

this issue is impacting the sound card and the "Billboard" device `with-interface 11:00:00` but is not impacting my USB mouse or USB keyboard which are both consistently authorized on resume.

khimaros, 2022-06-13:
after a system reinstall, this is still happening but now affecting the entire hub, which means all of the devices connected to the hub (including my keyboard and mouse) are not authorized after a suspend/resume.

khimaros, 2022-06-13:
i've started a thread with framework community and shared a (kludgy, possibly security impacting) workaround here: https://community.frame.work/t/usbguard-not-authorizing-devices-after-docked-suspend-resume/19449/2

lawa42, 2023-01-06:
Same here. Very annoying.