USBGuard/usbguard #557: [Feature Request] Kill switch functionality

PoisonFrog, 2022-09-24:
Really like this project but I would love if it also had a kill switch functionality that turns off the computer whenever an untrusted USB device is plugged in. I think this would help protect systems if an attacker gets physical access. I know that there scripts that do this, but I believe this would be a perfect fit for USBGuard as an optional feature.

muelli, 2022-11-10:
I don't see how it would fit the USBGuard model.
Currently, "all" it does is decide whether a device is allowed to be recognised by Linux.
No action, other than allowing Linux to further interact with the device, is taken.

PoisonFrog, 2022-11-10:
> I don't see how it would fit the USBGuard model. Currently, "all" it does is decide whether a device is allowed to be recognised by Linux. No action, other than allowing Linux to further interact with the device, is taken.

If an adversary plugs in an unauthorized device into a computer you own it might be better to have it automatically shut down and erase the encryption key from memory instead of allowing him more chances to temper with your device. At least for me it makes more sense to turn off the device instead of just blocking the USB to interact with the system and I don't see why USBGuard couldn't have both options.

muelli, 2022-11-15:
> I don't see why USBGuard couldn't have both options.

I haven't commented on the usefulness of the feature you proposed.
I have rather pointed out that USBGuard itself doesn't take any action. So your request entails acting on devices which I think is a relatively big thing to add. But it's Free Software! Go, write a patch! :)

You're probably aware of https://github.com/NateBrune/silk-guardian and https://github.com/hephaest0s/usbkill