USBGuard/usbguard #570: Add option --version

jk-10, 2022-12-01:
Running `usbguard --version` at least on Ubuntu 22.04 does not produce any versioning information. Handling these issues and fixes might be easier with known version numbers.

So please add support for new option `--version`

hartwork, 2022-12-01:
@jk-10 I second that there should be `--version` supported by at least command `usbguard`. :+1: 

Just for your information, one way to check which version of USBGuard (or any distro package) you have installed on Ubuntu (and Debian) is `apt-cache policy usbguard`, e.g.:
```console
# cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux bookworm/sid"
NAME="Debian GNU/Linux"
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

# apt-cache policy usbguard
usbguard:
  Installed: 1.1.2+ds-3+b1
  Candidate: 1.1.2+ds-3+b1
  Version table:
 *** 1.1.2+ds-3+b1 500
        500 http://deb.debian.org/debian sid/main amd64 Packages
        100 /var/lib/dpkg/status
```

Cropi, 2022-12-12:
I agree that this could come in handy when debugging future issues.

PR #576 (2024-05-30): Add "--version" option to the usbguard CLI
Running ```usbguard --version``` should help us handle fixes and issues more easily in the future. By doing so, it's easier to track down which options have been enabled during build time. Fixes #570

Any suggestions on what else to display are welcomed.
commit 8ec94ffc (2024-05-30): Add "--version" option to the usbguard CLI (#576)

* Add "--version" option to the usbguard CLI

Running usbguard --version should help us handle fixes and issues
more easily in the future. By doing so, it's easier to track down
which options have been enabled during build time. Fixes #570

* CI: replace deprecated include module in favor of include_tasks
closed 2024-05-30 by Cropi